labstreaminglayer/pylsl #2: StreamInlet.Time correction not working!

AdeuAndreu, 2018-11-27:
Hello guys, 

I am streaming EEG data using NIC software with ENOBIO. I see the stream of data both with matlab and python. 

My problem is that the StreamInlet object is not able to call the function time_correction. It does call but it last forever. I am placing a timeout of 60 seconds and timer expires before i can get a the time_correction calculated. Additionally, i am able to see my stream of data and also i am able to call pull_sample and pull_chunk. So, i don't know why it is not able to apply time_correction.

Do you know what it might be the cause of my error ? 

I am using python 3.6.6. I installed pylsl with pip and had not problems. In matlab it is able of calling time_correction without waiting too much.

Thank you a lot in advance!

Andrés

tstenner, 2018-11-27:
That's one of the trickier to find bugs we've had, but we've fixed it last week. It'll be in the next pypi release (@cboulay can we upload and mark experimental so they're not installed automatically?)

AdeuAndreu, 2018-11-28:
Thanks a lot for  your quick reply @tstenner I am very much looking forward to check the new version of your library. Do you have an expected date to upload it to the pypi server or to your git?

Cheers!

tstenner, 2018-11-28:
As a quick fix you can download the binary from https://github.com/sccn/liblsl/releases and replace the liblsl in the python package. We don't have an expected date, but we'll upload it as soon as we've done some tests and got some time.

AdeuAndreu, 2018-11-28:
Hello @tstenner , Thanks a lot for your message!
I followed your recommendation in the following way:
1. I downloaded the binary (liblsl-1.13.0-b2-Linux64-xenial.deb for linux)
2. i installed it -> which package manager does it in /usr/lib/. There is the liblsl64.so pointing to liblsl64.so.1.13.0 with a symbolic link
3. i copy the file liblsl64.so.1.13.0 to my python installation within my environment (envs/py36Env/lib/python3.6/site-packages/pylsl/)
4. i repalced the existing liblsl64.so with this the new binary(liblsl64.so.1.13.0)  keeping the liblsl64.so name

Things didn't breake and the library keeps working exactly the same. Unfortunately it does not calculate correctly the time_correction function still. It gets into an infinite waiting loop or just run the timeout. 

Did i do anything wrong ? Do you know what it might be the cause of the problem ?

LSL without time correction won't be useful for me cause i want to synchronize different streams.

I appreciate a lot your help! Thanks a lot in advance and have a great day!

Cheers!

cboulay, 2018-11-28:
Hi @AdeuAndreu , can you please provide us with a test case that doesn't depend on hardware? You can use one-to-several Python processes to push random data then another process to pull that data and trigger the bug you are seeing.

tstenner, 2018-11-29:
I have added a minimal example in d35c9ebda89bc7eee6d28a603f0ac92fadac4ac9. Could you also check with `python -c 'import pylsl; print(pylsl.pylsl.library_info())'` that in fact the correct liblsl is loaded?

AdeuAndreu, 2018-11-29:
Hi @cboulay , Thanks a lot for your message.
I am reading EEG data from a stream. i am able to pull_samples using the StreamInlet. It is very straight forward with the examples you provided. My problem is that i cannot run the time_correction. 

Apparently this bug is sorted out in the latest beta release. But I have not been able to well replace the binary or something is still missing.

I explained everything in my previous comment. 

Cheers!

tstenner, 2018-11-29:
> Output --> git:1.12-145-g5281edbf/build:/compiler:GNU-7.3.0/boost:1_55

That's still the old one, the string gets written to the liblsl binary at compilation time.

> -> when i imported pylsl i got the error OSERROR undefined symbol: _ZN8lslboost6system6detail20generic_category_ncxEv

That's very weird. I thought Python just loads the library and retrieves the function entry points afterwards, so no compilation step is involved. I'd just wait for the next pylsl release where the correct liblsl is included.

cboulay, 2018-11-29:
If you ‘git clone —recursive’ the full sccn/labstreaminglayer repo, then build liblsl, you should then be able to go into the lib liblsl-Python dir and do a ‘pip install .’

It’ll take a little while until I update pypi because I have to install a vm to properly build the ‘anylinux’ flavour.

AdeuAndreu, 2018-11-29:
@tstenner Thanks for you comment! I thinks it keeps a simbolic link somewhere, but it is my guess. 

@cboulay can you tell me the full address for the git repo?  Do i need to recursively clone https://github.com/labstreaminglayer/  or just the python-submodule? if i do so i will also download all the other staff for android and different submodules won't i ? 

If that does the trick, i can go on but just to verify. 

Cheers!

tstenner, 2018-11-29:
I've uploaded the package to my anaconda repository, if you're using conda you can install it via `conda install -c tstenner pylsl`

AdeuAndreu, 2018-11-29:
@tstenner thanks a lot for uploading the package to your anaconda repo. 
I created a new environment in python and i installed it pylsl again from scratch using conda install -c tstenner pylsl. Everything smooth until here.

When i run the commands:
import pylsl
print(pylsl.pylsl.library_info())

I get the output:
Output --> **git:/branch:/build:/compiler:GNU-7.3.0/boost:1_68**

Then:
I can pull_samples and pull_chunks. Then i call
 offset = inlet.time_correction(50) with 50 seconds timeOut.  

--> same problem going on. 
if you uploaded everything to anaconda repo with the latest beta  liblsl64.so release, i have not clue why it does not work. 

Cheers!

cboulay, 2018-11-29:
Sorry I was on my phone earlier and not able to type all the details.
If you don't already have a compiler, cmake, and boost, then you'll need to install those:

`sudo apt-get install build-essential cmake libboost-all-dev`

Then to clone the repo, build, install, then install pylsl:

`git clone --recursive https://github.com/sccn/labstreaminglayer.git`
`cd labstreaminglayer`
`mkdir build && cd build`
`cmake ..`
`cmake --build .`
`cd ..`
`cd LSL/liblsl-Python`
`pip install .`

AdeuAndreu, 2018-11-29:
@cboulay  thanks! i will try these steps then. I understand the last stept with pip must be within my python environment. right?  I will let you know how it goes. 

Cheers!

cboulay, 2018-11-29:
If you're talking in conda terms, then your environment has to be 'active'. i.e., `source activate my_env` before you pip install.

cboulay, 2018-12-17:
@AdeuAndreu You should be able to `pip install pylsl` and get the latest version. Tristan says your original issue was fixed in liblsl, and packages on pypi use the latest version of liblsl, so your issue should be solved with the latest version of pylsl from pypi. Please re-open this issue if the problem persists.

closed 2018-12-17 by cboulay

AdeuAndreu, 2019-01-11:
Hello guys! I retook the work i was doing with lsl and i want to thank your for uploading a new version with this small bug solved. The time correction is working fine for me with the new version. 

Cheers!