labstreaminglayer/pylsl #4: How to correctly match different data streams

AdeuAndreu, 2019-01-11:
Hello Guys,

First of all i apologise if this is not the right channel for posting this question as it is not a working issue, though practical. Let me know if I should use another channel. 

I am using LSL to read streaming of data from different devices and so far i am reading the the streams with the corrected timestamp. 

I want to create a unique dataset that match the data streams in time using the time stamps. So i would like to know how you do it. The problem that i have found is that the timestamps between streams don't fully match at millisecond level, which puts me off from doing the merge directly. Also another problem is that the sampling rate of the devices is different. So in a second you can have 500 samples from one device and 1000 or more from another. This also provides a another problematic to take into account.  

I know you have an app called labrecorder. So i though that you might have gone through this problem before and might know the best way to merge the data streams.

If you can tell me a hint that would be great!

Thanks a lot in advance!

Andrés

dmedine, 2019-01-11:
LabRecorder records native sampling rates only, and yes, you are right, 
the samples are not perfectly aligned on a grid. However, they are 
aligned in time. That is to say if you have stream A that has samples 
that occur at time 1 and 2, and stream B that has a sample that occurs 
at time .5, then that sample from stream B occurs half way between those 
samples from stream A.

agricolab is working on a Python solution for doing what you describe 
(https://github.com/sccn/xdf/pull/28). This will probably be merged into 
the main branch of XDF (the native LSL file format) soon.

AdeuAndreu, 2019-01-14:
Hello @dmedine 

Thanks for your quick answer. It was helpful to read the thread, the comments and to check some of the code for pyxdf. I found a way through very similar. I left a comment in that forum in case it could be useful.

Thanks a lot for your support!

Andrés

cboulay, 2020-01-11:
While this is a common issue, I think that resampling and aligning data is outside the scope of the LSL library itself. Based on the lack of activity in the past year, I assume others agree.

I like the idea of an app called "LSLAlignToGrid" that used post-processing on selected streams, buffering, resampling, aligning, then sending everything out in one giant stream. I'm sure this would be a welcome addition.

closed 2020-01-11 by cboulay