labstreaminglayer/pylsl #5: Errors in 1.13.1: Error during resolve_byprop: lslboost:: recursive_mutex constructor failed in pthread::cond_init

tatobi, 2019-01-16:
OS: Ubuntu 18.04
pylsl version: 1.13.1
Python in Conda:
```
root@machine:~# python
Python 3.5.5 | packaged by conda-forge | (default, Jul 23 2018, 23:45:43) 
[GCC 4.8.2 20140120 (Red Hat 4.8.2-15)] on linux
```
pip install -I pylsl==1.13.1

```
root@machine:~# python ReceiveData.py
looking for an EEG stream...
Error during resolve_byprop: lslboost:: recursive_mutex constructor failed in pthread::cond_init: Invalid argument
[]
Traceback (most recent call last):
  File "ReceiveData.py", line 16, in <module>
    inlet = StreamInlet(streams[0])
IndexError: list index out of range
root@engineering:~# 
```

__FIX:__

Remove new version, install 1.12.2:

```
pip uninstall pylsl
pip install -I pylsl==1.12.2
```

Then everything is fine, the data appearing in console.

cboulay, 2019-01-16:
To help narrow down the source of the problem, Can you please try building the latest liblsl from source and then pip install pylsl from source?

cboulay, 2019-01-16:
I was able to reproduce a similar problem using SendData.py: `Unexpected error during construction of stream outlet: lslboost::condition_variable::condition_variable() constructor failed in pthread::cond_init:`

Getting pip-installable binary-containing packages on Linux is a huge pain. They only accept manylinux wheels and these can only be built on super old distros. After great effort, @tstenner and I finally got things building and deploying but I guess these packages aren't in fact 'manylinux' compatible.

We might have to remove pip-support and rely exclusively on conda for Linux. It will be a while until we have conda packages on conda-forge. Until then, your best bet is to build and install pylsl yourself.

cboulay, 2019-01-16:
Even though your issue isn't resolved, I'm going to close this issue and move the conversation to #6 which lays out the problem and some possible solutions.

closed 2019-01-16 by cboulay