labstreaminglayer/pylsl #10: [Question] Send and Receive Stream on same computer?

teonbrooks, 2019-04-04:
Hi, I am working on building LSL for MNE-Python and I am doing some local testing. I was wondering if it is possible to both create a stream and read from the stream from the same computer. Currently I am creating a mock stream using the example and I am trying to consume it in another Python instance.

cboulay, 2019-04-04:
Yes that should be no problem. Are you having any trouble?

BTW, for your use case, I want you to be aware of something: `pip install pylsl` on Linux gives an older version of pylsl that comes packaged with a bunch of .dll, .dylib and .so files, with the .so only working on a couple versions of debian/ubuntu. I'm waiting for manylinux2010 to stabilize before I provide a proper updated release for pylsl on Linux. Until then, on Linux you're better of doing `conda install -c tstenner pylsl`.

teonbrooks, 2019-04-04:
thanks for the quick response @cboulay. I'm getting this error message when I try to subscribe to the StreamInlet:
```python
python(42387,0x110dd25c0) malloc: can't allocate region
*** mach_vm_map(size=3761460517889732608) failed (error code=3)
python(42387,0x110dd25c0) malloc: *** set a breakpoint in malloc_error_break to debug
Unexpected error during construction of a stream_inlet: std::bad_alloc
```

teonbrooks, 2019-04-04:
I'm working on a Mac, I used `pip` today to install pylsl

cboulay, 2019-04-04:
What version of MacOS are you running? Are you creating the inlet within some mne- or Qt-thread?

I'll be in front of a macbook tomorrow and be able to try then. If you'd rather not wait, you can try to build and install pylsl yourself following the 'self-built' instructions on the README. If that still fails, then you'll have to `git clone --recursive https://github.com/sccn/labstreaminglayer.git`, build liblsl yourself, then try to install pylsl. At least those will be the steps I'll take tomorrow if `pip install pylsl` fails for me.

teonbrooks, 2019-04-04:
I'm using MacOS Mojave. I constructed a [Client](https://github.com/teonbrooks/mne-python/blob/rtclient/mne/realtime/lsl_client.py) class to fit in with the way MNE handles realtime data. https://github.com/mne-tools/mne-python/pull/6120 has the latest of what I've been working on.

```python
import mne; 
with mne.realtime.LSLClient('myuid34234') as client:
    epoch = client.get_data_as_epoch()
```

i've also tried it outside of the Class I created directly with pylsl and it gives the same error. 

> I'll be in front of a macbook tomorrow and be able to try then.

cool. headed to bed now. will look into building liblsl myself tomorrow if that seems to be the solution

cboulay, 2019-04-04:
Everything looks like it's working on my end. I'm on MacOS Mojave 10.14.3.
I'm using a conda env with python 3.7. I did `pip install pylsl` and that seemed to work well.

`python -c "import pylsl; print(pylsl.library_info())"` gives `git:1.13.0-b3-24-g242ef5f8/branch:master/build:/compiler:AppleClang-9.1.0.9020039/boost:1_69`

`python -c "import pylsl; print(pylsl.library_version())"` gives `113`.

In one terminal I run `python -m pylsl.examples.SendStringMarkers`, and in another terminal with the same conda env activated I run `python -m pylsl.examples.ReceiveStringMarkers` and get output like the following...

```
got XXX at time 531560.127594056
got Test-1-2-3 at time 531560.808666021
got XXX at time 531561.431121153
got Marker at time 531562.149851102
got Test-1-2-3 at time 531563.585715256
```

tstenner, 2019-04-04:
> git clone --recursive https://github.com/sccn/labstreaminglayer.git, build liblsl yourself, then try to install pylsl. At least those will be the steps I'll take tomorrow if pip install pylsl fails for me.

Quick addendum: you can clone the liblsl and liblsl-Python repositories separately and build both packages for conda with `conda build path/to/liblsl; conda build path/to/liblsl-Python`.

teonbrooks, 2019-04-04:
ah, found my error. I forgot that resolve_byprop yields a list of StreamInfo so I wasn't correctly creating the inlet. thanks @cboulay for the snippets for the examples. Those are really helpful. I will close this issue

closed 2019-04-04 by teonbrooks