labstreaminglayer/pylsl #12: Pylsl precision problems

nmc-costa, 2019-04-15:
Below you can see a test of outlet and inlet data. Only 5 significant numbers are equal. Is this normal?

**********************OUTLET_DATA*******************
STAMP:[17179.395466097]
SAMPLE:[52.057985450431254, 47.2703647175461, 15.023354318885762, 56.304367890891285, 42.92836237812874, 25.00853755102409, 26.02998306266926, 40.27109700121555, 4.996895512545263, 33.472730456238885, 39.705453031294994, 84.91143736067775, 62.38916076898047, 27.758021899765893, 51.15501291708292, 89.9828961474747, 33.95563917840357, 31.503025009413797, 45.33956661540789, 51.09868060197509, 21.734038834742563, 71.64371584605918, 86.05013294594738, 61.075437121788035, 13.544131172100393, 10.255098534937835, 39.304440040989775, 31.73323912424252, 19.11649928209035, 69.97309570675311, 24.876553585727113, 95.42718716719125]

**********************INLET_DATA*******************
STAMP:[17179.395466097]
SAMPLE:[52.057987213134766, 47.270362854003906, 15.023354530334473, 56.30436706542969, 42.92836380004883, 25.00853729248047, 26.029983520507812, 40.271095275878906, 4.9968953132629395, 33.47272872924805, 39.70545196533203, 84.91143798828125, 62.38916015625, 27.75802230834961, 51.15501403808594, 89.98289489746094, 33.95563888549805, 31.50302505493164, 45.33956527709961, 51.09868240356445, 21.734039306640625, 71.64371490478516, 86.05013275146484, 61.075435638427734, 13.5441312789917, 10.255098342895508, 39.304439544677734, 31.733238220214844, 19.116498947143555, 69.97309875488281, 24.876554489135742, 95.42718505859375]

CODE USED FOR PRINTING THE DATA
outlet.py:
from pylsl import StreamInfo, StreamOutlet, local_clock
sampleRate = 1000
channel_number = 32
info = StreamInfo('Actichamp', 'EEG', channel_number, sampleRate, 'float32', 'myuid2424')
outlet = StreamOutlet(info, 10, 360)
start_time=local_clock() #s
elapsed_time=local_clock()-start_time #s
while elapsed_time<=100:
    sample=[]
    for i in range(channel_number):
        sample.append(random.random())
    offset = 100 
    m_array = np.array(sample)*offset
    sample = m_array.tolist()
    sample_stamp = [local_clock()-0.125]
    outlet.push_sample(sample, sample_stamp[0])
    print('**********************OUTLET_DATA*******************')
    print('STAMP:'+str(sample_stamp))
    print('SAMPLE:'+str(sample))

    time.sleep(1/float(sampleRate))
    elapsed_time=local_clock()-start_time

inlet.py:
from pylsl import StreamInlet, resolve_stream
print("looking for an EEG stream...")
streams = resolve_stream('type', 'EEG')
inlet = StreamInlet(streams[0])
while True:
    chunk, timestamps = inlet.pull_chunk()
    if timestamps:
        print('**********************INLET_DATA*******************')
        print('SAMPLE_STAMP:'+str(timestamps))
        print('SAMPLE:'+str(chunk))

Cheers,
NC

agricolab, 2019-04-15:
>  Is this normal?

I'd say, yes

```
outlet = 52.057985450431254
inlet =  52.057987213134766
outlet-inlet
>>> -1.7627035120426626e-06
```
Six significant decimal digits are within the specifications of floats. See https://en.wikipedia.org/wiki/Single-precision_floating-point_format. If you really need more precision, you might wanna try double precision floats or integer encoding.

tstenner, 2019-04-16:
Also in plain Python:

```
>>> np.array([52.057985450431254], dtype=np.float32)
array([52.057987], dtype=float32)
```

closed 2019-04-16 by tstenner