labstreaminglayer/pylsl #16: Pylsl on Nvidia Jetson Nano with Ubuntu Image

ShawhinT, 2019-07-10:
Hello all, 

I apologize in advance if my post is improper. I am a bit of an outsider to the software development world. I am trying to use pylsl on an Nvidia Jetson Nano running an Ubuntu 18.04 image. I have run the following:

`pip install pylsl`

with the output:

> Collecting pylsl
>  Using cached https://files.pythonhosted.org/packages/a9/63/ffa84bb429d1b42a7b3451e11847cbfb95a2151ed4016c99e1ba732f7252/pylsl-1.12.2-py2.py3-none-any.whl
> Installing collected packages: pylsl
> Successfully installed pylsl-1.12.2

Which seems fine to me. Then when trying to import in python I get the error:

> Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>  File "/home/teamlary/.local/lib/python2.7/site-packages/pylsl/__init__.py", line 2, in <module>
>  from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
> File "/home/teamlary/.local/lib/python2.7/site-packages/pylsl/pylsl.py", line 1200, in <module> 
> lib = CDLL(libpath)
> File "/usr/lib/python2.7/ctypes/__init__.py", line 366, in __init__
> self._handle = _dlopen(self._name, mode)
> OSError: /home/teamlary/.local/lib/python2.7/site-packages/pylsl/liblsl64.so: cannot open shared object file: No such file or directory

I also get an identical issue when using pip3 and python3. Has anyone encountered a similar issue? Any suggestions on the best way to get this working would be very much appreciated!

Cheers,

Shawhin Talebi
Underpaid Research Assistant

tstenner, 2019-07-11:
Hi,

the Python package is a wrapper around the liblsl.so binary which has to be compiled for each architecture (arm7?) and OS ([see here](https://github.com/sccn/labstreaminglayer/blob/master/doc/BUILD.md#note-about-architectures--binaries)). So, on your Ubuntu image you'll need to compile a liblsl:

```
sudo apt install cmake gcc
git clone https://github.com/sccn/liblsl
cd liblsl
mkdir build
cd build
cmake ..
cmake --build . --target install
```

After that, you'll find the `liblsl64.so` in `liblsl/build/install/lib/`.

ShawhinT, 2019-07-11:
tstenner, you are a wonderful human being, thank you! I will give that a try and give an update here.

ShawhinT, 2019-07-12:
It works, thank you tstenner! A couple notes for anyone with a similar issue.

1. I had to install cmake and gcc seperately i.e.

`sudo apt install cmake`
`sudo apt install gcc`

2. Then I had to move the shared object files to a directory like:

> ~/.local/lib/python3.6/site-packages/pylsl/

or

> ~/.local/lib/python2.7/site-packages/pylsl/

Thanks again!

closed 2019-07-12 by ShawhinT

cboulay, 2019-07-12:
For step 2,

* you can clone this repo, copy the built lib folder into the cloned folder, then do `pip install .`
* you can manually copy the shared objects to another search location.
    * I recommend `/usr/local/lib` so then other applications or Python environments will have access to it.
    * Alternatively, next to the pylsl.py file like you did is a good option too, but will have to be done separately for each Python environment that uses pylsl.

ShawhinT, 2019-07-12:
That is good to know. Thanks cboulay!

tstenner, 2019-07-14:
Or, even better, you can run `cpack -G deb` from the `build` directory and den install the `liblsl-…-amd64.deb` so the package manager can take care of it.

abhincho, 2021-07-02:
Hi
I am also facing the same problem. I followed the steps suggested by @tstenner. However in my case only the liblsl.so
files are getting generated and not the /liblsl64.so. Please help

Thank you