labstreaminglayer/pylsl #23: Recording marker stream with lab recorder

TassLabNeurofeedback, 2020-02-25:
Hello,

I am having trouble recording a marker stream written in python along with the output stream. When I open the xdf file in eeglab, all of the markers are bunched up at the end of the recording. I know it's not my code, because this happens even for the example code you provide. The one thing that is the same is that I am testing this on .set data streaming over neuropype. Any idea what my issue might be? Thanks.

-Alex

cboulay, 2020-02-25:
Hi Alex,
I'm sorry I didn't fully understand your question. I don't understand how the '.set data streaming over neuropype' plays into this. Can you break it down a little more? Or maybe hop on to Slack and chat with me there? You can find the Slack invite link [on this page](https://labstreaminglayer.readthedocs.io/info/getting_started.html#getting-help).

cboulay, 2020-02-26:
The problem was a parameterization of the the NeuroPype node. In NeuroPype, using the `LSLOutput` node defaults to using the time stamps in the data, but the markers being sent by the Python script were using the current timestamps from the LSL clock, so when combined the data was very old and the timestamps very new. The `LSLOutput` node has an argument `use_data_timestamps` which can be set to False to use the current LSL clock.

closed 2020-02-26 by cboulay