labstreaminglayer/pylsl #25: Having problems which I never encountered before. Please help.

Rakesh420, 2020-08-06:
Traceback (most recent call last):
  File "1.py", line 1, in <module>
    import pylsl
  File "C:\Users\rakes\anaconda3\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\rakes\anaconda3\lib\site-packages\pylsl\pylsl.py", line 1217, in <module>
    lib = CDLL(libpath)
  File "C:\Users\rakes\anaconda3\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'C:\Users\rakes\anaconda3\lib\site-packages\pylsl\lib\liblsl64.dll' (or one of its dependencies). Try using the full path with constructor syntax.

cboulay, 2020-08-06:
Hi Rakesh. I'm sorry but based on your question I don't have enough information to help you.

> "Having problems which I never encountered before".
Before what? What have you done differently since before?

What are you trying to do?

How did you install pylsl? When did you install pylsl? What version does the package manager think it is?

Rakesh420, 2020-08-06:
I am not able to import the pylsl package. I installed the pylsl using pip install pylsl. I always did the same and never had problems.

Rakesh420, 2020-08-07:
I had installed the latest version of python as I migrated to new computer. I reverted back to python 3.6 and the problem doesn't occur now.

closed 2020-08-07 by Rakesh420

cboulay, 2020-08-07:
I just tried with python 3.7 and 3.8 without issue. And I tried to get conda to install python 3.9 but there is no easy way, so I don't think that's worth investigating now.
If you can reproduce your original error then please let us know.

Rakesh420, 2020-08-07:
I will try to reproduce the error again. I uninstalled conda and installed
python 3.6 and it was working fine. Maybe I had problems with anaconda.