labstreaminglayer/pylsl #26: Problem importing pylsl in Windows 10 (Python-3.8.5)

giorgioganis, 2020-08-07:
Hello, I'm trying to do some tests combining pygame and pylsl using Windows 10 for some simple neurofeedback demos. 

I installed pylsl using pip version 20.2.1 ( python3 -m pip install pylsl ). This seems to install pylsl successfully:

_Name: pylsl
Version: 1.13.6
Summary: Python interface to the Lab Streaming Layer
Home-page: https://github.com/labstreaminglayer/liblsl-Python
Author: Christian Kothe
Author-email: christiankothe@gmail.com
License: MIT
Location: c:\users\gganis\python38-32\lib\site-packages
Requires:
Required-by:_

However, when I try to import pylsl within the Python3 shell I get this error:

_Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\gganis\Python38-32\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\gganis\Python38-32\lib\site-packages\pylsl\pylsl.py", line 1222, in <module>
    lib.lsl_library_info.restype = c_char_p
  File "C:\Users\gganis\Python38-32\lib\ctypes\__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\gganis\Python38-32\lib\ctypes\__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'lsl_library_info' not found_

Any thought on what might be off? Thanks.

cboulay, 2020-08-07:
This is the 3rd such complaint in as many days, so there's something going on, but I don't know what it is and I haven't been able to replicate it.

What seemed to work in those other cases was to uninstall & reinstall pylsl or the whole python environment.
I wonder if pypi is having a bad week?

(by the way, I don't know if you intended to but you're using 32-bit python which is a little bit unusual these days. I used 32-bit below to check, but I also checked 64-bit which also works fine)

```
>set CONDA_FORCE_32BIT=1
>conda create -n py38_32 python=3.8
>conda activate py38_32
>pip install pylsl
>python
```
> Python 3.8.5 (default, Aug  5 2020, 01:17:43) [MSC v.1916 32 bit (Intel)] :: Anaconda, Inc. on win32
```
>>> import pylsl
>>> pylsl.library_info()
'git:/branch:/build:/compiler:MSVC-16.0.40219.1/boost:1_69'
```

Where did you get your Python? (I notice MSC v.1926 for you vs 1916 for me).

cboulay, 2020-08-07:
I found the same version of Python --> no error
I then updated pip --> error!

I'll work on it.

cboulay, 2020-08-07:
When I try to force pip to install the [correct wheel that I download manually from pypi](https://files.pythonhosted.org/packages/ed/5a/f9fb83b288459eca12cecdf3be810e18f3a52681ec91830ab4dcf4a0d87d/pylsl-1.13.6-py2.py3-none-win32.whl) -- this is the same wheel that gets installed properly in conda when it is set to force 32bit -- pip reports the following:

> ERROR: pylsl-1.13.6-py2.py3-none-win32.whl is not a supported wheel on this platform.

Worse yet, it's totally happy to install the amd64 wheel.

Somehow pip coupled with Python 3.8 32-bit (the default downloaded from python.org) is trying to install 64-bit wheels and refuses to install 32-bit wheels. I confirmed this with another package. I even tried another package (numba) and same thing: wouldn't install 32-bit wheel but would install 64-bit wheel.

You can hack around this by replacing the `-win32` in the filename with `-any`, then pip will happily install it.

This is a problem with Python and pip. I can't do anything about it. But I do suggest that you stop using 32-bit python if you can. Some packages don't even support 32-bit Python anymore so you're limiting yourself. Unfortunately, Python.org's default download link gives you 32-bit Python but that's mostly for historical reasons. You should pass the button and follow through to the link: https://www.python.org/downloads/windows/
Or, better yet, use `miniconda`: https://docs.conda.io/en/latest/miniconda.html

closed 2020-08-07 by cboulay