labstreaminglayer/pylsl #27: Having trouble importing pylsl

devaka40, 2020-08-10:
Hi everyone, I having trouble importing pylsl. Everytime I try to import it I get:

Traceback (most recent call last):
File "C:/Users/computer/PycharmProjects/pythonProject1/MuseRead.py", line 33, in
import pylsl
File "C:\Users\computer\PycharmProjects\pythonProject1\venv\lib\site-packages\pylsl_init_.py", line 2, in
from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,
File "C:\Users\computer\PycharmProjects\pythonProject1\venv\lib\site-packages\pylsl\pylsl.py", line 1217, in
lib = CDLL(libpath)
File "C:\Users\computer\AppData\Local\Programs\Python\Python37\lib\ctypes_init_.py", line 364, in init
self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

I'm tried looking at the installation guide for pylsl and the only thing I can see is to use pip install pylsl which is what I used.

Thanks in advance to everyone that helps.

cboulay, 2020-08-10:
Are you using 32-bit Python? Can you switch to 64-bit?

devaka40, 2020-08-10:
Thanks, but currently using 64-bit Python.

I run the following to check it.

 import sys
>>> print("%x" % sys.maxsize, sys.maxsize > 2**32)
7fffffffffffffff True

cboulay, 2020-08-10:
How are you managing your Python environment? I don't think I've ever seen `site-packages\pylsl_init_.py`.

This error either happened because the wheel file got corrupted during download or unpacking, or there's a 32-64 mismatch between your Python interpreter, the tool you're using to manage your environment, and/or the ctypes module.

If you let me know exactly how you've created your python environment and installed pylsl (including every detail like what command prompt you opened) then I will try to replicate it locally.

devaka40, 2020-08-10:
I'm using the default PyCharm environment manager and used the built in package installer. I have also tried to activate the environment in the command line and uninstalling the pylsl and reinstalling it using pip (with the cache and without). 
For installing pylsl using cache:
(venv) C:\Users\computer\PycharmProjects\pythonProject1\venv\Scripts>pip install pylsl
Collecting pylsl
  Using cached pylsl-1.13.6-py2.py3-none-win32.whl (336 kB)
Installing collected packages: pylsl
Successfully installed pylsl-1.13.6
For uninstalling pylsl:
(venv) C:\Users\computer\PycharmProjects\pythonProject1\venv\Scripts>pip uninstall pylsl
Found existing installation: pylsl 1.13.6
Uninstalling pylsl-1.13.6:
  Would remove:
    c:\users\computer\pycharmprojects\pythonproject1\venv\lib\site-packages\pylsl-1.13.6.dist-info\*
    c:\users\computer\pycharmprojects\pythonproject1\venv\lib\site-packages\pylsl\*
Proceed (y/n)? y
  Successfully uninstalled pylsl-1.13.6
For installing without using cache:
(venv) C:\Users\computer\PycharmProjects\pythonProject1\venv\Scripts>pip install pylsl --no-cache-dir
Collecting pylsl
  Downloading pylsl-1.13.6-py2.py3-none-win_amd64.whl (363 kB)
     |████████████████████████████████| 363 kB 3.2 MB/s
Installing collected packages: pylsl
Successfully installed pylsl-1.13.6

cboulay, 2020-08-10:
From your log you can see that the first time it installed 32-bit version, then the second time it installed 64-bit version. Is there any difference in the result?

devaka40, 2020-08-10:
Same results.

devaka40, 2020-08-10:
(venv) C:\Users\computer\PycharmProjects\pythonProject1\venv\Scripts>python
Python 3.7.8 (tags/v3.7.8:4b47a5b6ba, Jun 28 2020, 08:53:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylsl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\computer\PycharmProjects\pythonProject1\venv\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\computer\PycharmProjects\pythonProject1\venv\lib\site-packages\pylsl\pylsl.py", line 1217, in <module>
    lib = CDLL(libpath)
  File "C:\Users\computer\AppData\Local\Programs\Python\Python37\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

cboulay, 2020-08-10:
I'm using PyCharm Professional 2020.1.2.
I created a new virtual environment. I used the GUI to install pylsl. I opened a console and did `import pylsl` and `pylsl.library_info()` and they both worked as expected. A big difference though is my 'base interpreter' was from my miniconda install which was 3.6. I repeated the process with a different conda base of 3.7 (``) and again no errors.

Can you verify things work normally for you if you use a conda environment? https://docs.conda.io/en/latest/miniconda.html

cboulay, 2020-08-10:
Sorry I just realized something else. In your traceback, it seems to be using 2 different environments. One in PycharmProjects\pythonProject1\venv and another in AppData\Local\Programs\Python\Python37. The latter is where ctypes is being invoked, and this environment probably 32-bit somehow and/or mismatches in another way.

I'm not going to be able to help you any further. I think you need to nuke all your python environments and start over.

devaka40, 2020-08-10:
Thank you very much for your time and help.

cboulay, 2020-08-11:
Were you able to solve it? Please post here when you do. I'll go ahead and close the issue because there's nothing we can do about it.

closed 2020-08-11 by cboulay