labstreaminglayer/pylsl #30: liblsl 1.14 and python 3.8 -- Exception on creating XML element.

cboulay, 2020-09-15:
Python 3.8.5 installed via conda, using pylsl wheel built with liblsl 1.14b03

```Python
import pylsl
info = pylsl.StreamInfo(channel_count=3, channel_format=pylsl.cf_string)
chans = info.desc().append_child("channels")
```

> Unhandled exception in event loop:
  File "C:\Users\Chad\.conda\envs\pylsl114\lib\asyncio\proactor_events.py", line 768, in _loop_self_reading
    f.result()  # may raise
  File "C:\Users\Chad\.conda\envs\pylsl114\lib\asyncio\windows_events.py", line 808, in _poll
    value = callback(transferred, key, ov)
  File "C:\Users\Chad\.conda\envs\pylsl114\lib\asyncio\windows_events.py", line 457, in finish_recv
    raise ConnectionResetError(*exc.args)

> Exception [WinError 995] The I/O operation has been aborted because of either a thread exit or an application request
Press ENTER to continue...

I swapped to python 3.7, used the exact same wheel, and couldn't reproduce the issue.

tstenner, 2020-09-16:
```
> python
Python 3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylsl
>>> info = pylsl.StreamInfo(channel_count=3, channel_format=pylsl.cf_string)
>>> chans = info.desc().append_child("channels")
>>> pylsl.library_info()
'git:v1.14.0b3-1-g2a0f6ba2/branch:master/build:RelWithDebInfo/compiler:MSVC-19.28.29304.1/link:shared'
```

cboulay, 2020-09-16:
Yeah I can't reproduce it now either.
I'll blame it on my computer being in a bad state for whatever reason. I admit that I didn't try turning my computer off then on again. I should have.

closed 2020-09-16 by cboulay

tstenner, 2020-09-16:
https://xkcd.com/1987/