labstreaminglayer/pylsl #31: liblsl64.dll cannot be found

Hororohoruru, 2020-09-16:
Hello, I am trying to make pylsl work on Windows 7, using Python 3.8.5.

When import pylsl I get the follosing traceback:

```
In [1]: import pylsl
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-1-d9db44c5cf92> in <module>
----> 1 import pylsl

c:\users\nirx1\anaconda3\envs\psychopy\lib\site-packages\pylsl\__init__.py in <m
odule>
      1 # (not using import * for Python 2.5 support)
----> 2 from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float3
2,\
      3     cf_double64, cf_string, cf_int32, cf_int16, cf_int8, cf_int64,\
      4     cf_undefined, protocol_version, library_version, library_info, local
_clock,\
      5     proc_ALL, proc_none, proc_clocksync, proc_dejitter, proc_monotonize,
 proc_threadsafe,\

c:\users\nirx1\anaconda3\envs\psychopy\lib\site-packages\pylsl\pylsl.py in <modu
le>
   1215                        "subfolder of the pylsl package or the system sea
rch "
   1216                        "path). Alternatively, specify the env var PYLSL_
LIB")
-> 1217 lib = CDLL(libpath)
   1218
   1219 # set function return types where necessary

c:\users\nirx1\anaconda3\envs\psychopy\lib\ctypes\__init__.py in __init__(self,
name, mode, handle, use_errno, use_last_error, winmode)
    379
    380         if handle is None:
--> 381             self._handle = _dlopen(self._name, mode)
    382         else:
    383             self._handle = handle

FileNotFoundError: Could not find module 'c:\users\nirx1\anaconda3\envs\psychopy
\lib\site-packages\pylsl\lib\liblsl64.dll' (or one of its dependencies). Try usi
ng the full path with constructor syntax.
```

Then I go to check the path and liblsl64.dll is indeed on the specified path, on the lib\ folder, so I don't know what else to include in order for the library to work. I installed pylsl via pip.

cboulay, 2020-09-16:
You can try this pylsl I just built, attached to this post. Unzip then `pip install {name_of_wheel}`
[pylsl-1.14.0-py2.py3-none-win_amd64.zip](https://github.com/labstreaminglayer/liblsl-Python/files/5232649/pylsl-1.14.0-py2.py3-none-win_amd64.zip)

Even if that works, it would be great if you'd be willing to help figure out what's missing on your system. Please try the following:
* Download and unpack [Dependencies](https://github.com/lucasg/Dependencies/releases)
* From an anaconda prompt with your psychopy environment activated, run `DependenciesGui.exe`
* Use File->Open to navigate to and select liblsl64.dll
* See if any of the dependencies in the tree appear in red, indicating that they cannot be found. If so then please take a screenshot of that and attach it here.

If one of the missing dependencies looks to be related to microsoft runtimes (e.g. MSV140.dll), then you can try downloading and installing vc_redist.x64.exe from [here](https://support.microsoft.com/en-ca/help/2977003/the-latest-supported-visual-c-downloads).

Hororohoruru, 2020-09-16:
Hello. Thank you for your answer! The .whl file did not fix the issue. I proceeded with your indications and found 2 missing dependencies. Here you can find a screenshot 
[image]

The ...140_1.dll looks like something that a lot of people is missing from the C++ redistributables, so I will look into that.  The other one that appears in red, I don't know.

Interestingly, when I tried to import pylsl from the native Python distribution in my computer (not conda) if worked. Any insights from this?

Again, thank you very much.

cboulay, 2020-09-16:
I think the WS2_32.dll is a false positive. That file is there. If you expand it (click on arrow), you'll probably see that only 1 of its dependencies is missing and it's something obscure that doesn't matter.

But missing VCRUNTIME140_1.dll is a problem. My guess would be that the native Python distribution ships with its own VCRUNTIME140_1.dll that the conda distribution is somehow missing. I think if you install vc_redist.x64.exe from above then it will fix the problem for you. If you don't want to install that at the system level, there's probably a way to get conda to install it, maybe `conda install vs2015_runtime`?

cboulay, 2020-09-18:
Did the vc_redist work for you? If yes then please close this issue.

tstenner, 2020-09-18:
The problem is that since MSVC2015 the runtime library (`VCRUNTIME140*`) are backwards compatible, but not forward compatible, so the MSVC2017 runtime (`VCRUNTIME140_1.dll`) is needed. Since they're backwards compatible, Microsoft only distributes the MSVC2015-2019 runtime.
This needs to be documented better, preferable with the error message.

Hororohoruru, 2020-09-18:
Hello. Installing both the x64 and x86 versions worked. I had to:

- Uninstall any C++ redistributables present from the range 2015-2019
- Run the .exe files provided [here](https://support.microsoft.com/en-ca/help/2977003/the-latest-supported-visual-c-downloads)

Thank you very much for your help!

closed 2020-09-18 by Hororohoruru