labstreaminglayer/pylsl #32: OSError: [WinError 193] %1 is not a valid Win32 application

rakeshbci, 2020-09-18:
from pylsl import StreamInlet, resolve_stream
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python38\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python38\lib\site-packages\pylsl\pylsl.py", line 1217, in <module>
    lib = CDLL(libpath)
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python38\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

cboulay, 2020-09-18:
Did you get Python from python.org? I noticed the default 3.8 download is for 32-bit Python, but that 32-bit Python somehow installs 64-bit wheels which are incompatible. That might be the case here.

More likely is that you're missing a visual studio runtime. Download and install vc_redist.x64 from [here](https://support.microsoft.com/en-ca/help/2977003/the-latest-supported-visual-c-downloads).

rakeshbci, 2020-09-18:
I uninstalled my python and installed python 3.7.9 and reinstalled the vc_redist.x64 too. I am still unable to run the script. This was working fine the last week. 
C:\Rakesh work\programming workspace\Finalproject\drone blocks>python --version
Python 3.7.9

C:\Rakesh work\programming workspace\Finalproject\drone blocks>pip install pylsl --no-cache-dir
Collecting pylsl
  Downloading pylsl-1.13.6-py2.py3-none-win_amd64.whl (363 kB)
     |████████████████████████████████| 363 kB 66 kB/s
Installing collected packages: pylsl
Successfully installed pylsl-1.13.6
WARNING: You are using pip version 20.1.1; however, version 20.2.3 is available.
You should consider upgrading via the 'c:\users\rakes\appdata\local\programs\python\python37\python.exe -m pip install --upgrade pip' command.

C:\Rakesh work\programming workspace\Finalproject\drone blocks>python 1.py
Traceback (most recent call last):
  File "1.py", line 3, in <module>
    from pylsl import StreamInlet, resolve_stream
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python37\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python37\lib\site-packages\pylsl\pylsl.py", line 1217, in <module>
    lib = CDLL(libpath)
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python37\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

agricolab, 2020-09-18:
I have received (not for pylsl, but for other dlls) similar errors on Windows, although i knew the file was there and the path i specified was correct. In my experience, this usually is a windows or environment bug, and tricky to resolve. I once eventually resorted to brute force, i.e. i loaded the dll in a while-loop until it was successfully loaded. Don't ask me why that worked. 

In your case, maybe the search path is messed up from earlier installs? If there is still a path to an old python installation, the new one might be shadowed and you might need to clean your path. You can print the current PATH from `cmd` with
`echo %PATH:;=&echo.%` 

Can you also confirm that you are using a [64bit install](https://stackoverflow.com/questions/1405913/how-do-i-determine-if-my-python-shell-is-executing-in-32bit-or-64bit) of python?

rakeshbci, 2020-09-18:
Yes, I am using 64bit python.

cboulay, 2020-09-18:
It's almost certainly an architecture mismatch problem, either that or pylsl 1.13.6 is somehow corrupted on pypi. [This SO answer](https://stackoverflow.com/a/57297745) has a lot of relevant information.

You can try this newer wheel: 
[pylsl-1.14.0-py2.py3-none-win_amd64.zip](https://github.com/labstreaminglayer/liblsl-Python/files/5246989/pylsl-1.14.0-py2.py3-none-win_amd64.zip)

Or you can try an older wheel with `pip install pylsl==1.13.1`.

rakeshbci, 2020-09-18:
This did not solve my problem. I got this output when I tried installing from the .whl and with the older package too.
from pylsl import StreamInlet, resolve_stream
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python37\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python37\lib\site-packages\pylsl\pylsl.py", line 1200, in <module>
    lib = CDLL(libpath)
  File "C:\Users\rakes\AppData\Local\Programs\Python\Python37\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

cboulay, 2020-09-18:
You've somehow managed to break your environment. There are some incorrect dlls on some path that are shadowing the correct dlls... or something like that.

I recommend uninstalling all of your manually installed python interpreters and switch to using conda-managed environments. I like to use [miniconda](https://docs.conda.io/en/latest/miniconda.html).

rakeshbci, 2020-09-21:
I manually installed python and removed all the path from the Enovironmental variables. I installed minicoda and still I am getting the same error.
File "1.py", line 3, in <module>
    from pylsl import StreamInlet, resolve_stream
  File "C:\Users\rakes\miniconda3\envs\lsl\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\rakes\miniconda3\envs\lsl\lib\site-packages\pylsl\pylsl.py", line 1217, in <module>
    lib = CDLL(libpath)
  File "C:\Users\rakes\miniconda3\envs\lsl\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

cboulay, 2020-09-21:
Please download and unpack [Dependencies](https://github.com/lucasg/Dependencies/releases) somewhere.
Inside your anaconda prompt window, with your `lsl` env active, run `path\to\DependenciesGui.exe`. In the DependenciesGui, click file-->open and find the dll inside `C:\Users\rakes\miniconda3\envs\lsl\lib\site-packages\pylsl\lib\`

Please list here the dependencies in red that are missing. If possible, expand a missing item to drill down to _its_ missing dependency.

rakeshbci, 2020-09-22:
[image]
[image]
[image]
[image]
[image]

cboulay, 2020-09-23:
I think those files that Dependencies reports as missing are false positives due to localization. See here: https://github.com/lucasg/Dependencies/issues/102

So... I have no idea what the problem is. You can try `pip install pylsl==1.13.1`. That one comes with an older DLL that is more compatible.

rakeshbci, 2020-09-24:
I installed the pylsl 1.13.1, I still have the dependencies problem.

cboulay, 2020-09-24:
I really have no idea what's wrong with your system. Do you have access to another computer running the same (localized) version of Windows that you can try? Or can you ask a friend to try?
Assuming that other computer works, then my advice is to reinstall windows. I'm sorry we couldn't pinpoint the problem.

agricolab, 2020-09-24:
I agree with @cboulay that this is likely an environment issue, and not caused by the liblsl.dll. The dll obviously exists, and runs fine on other machines. My last shot: Can you try a brute-force approach? I once had brute-force work on windows for a different dll, so maybe it is worth the try. If it ran for a few minutes unsucessfully, just abort with ctrl-c.
``` python
print("[", end="")
while True:
    try:
        print(".", end="")
        from pylsl import StreamInlet, resolve_stream
    except OSError as e:
        pass
    else:
        print("]")
        break

```

tstenner, 2020-09-24:
Just to make sure pylsl tries to load the right library: you can set the environment variable [`PYLSL_LIB`](https://github.com/labstreaminglayer/liblsl-Python/blob/349d5becd0e92600946379874635b77a775a55a1/pylsl/pylsl.py#L1185) to a full path; pylsl then tries to load this file as lsl library.

rakeshbci, 2020-09-29:
@tstenner I added the PYLSL_LIB as a new environment variable and set it to the full path. This solved my problem. Thank you so much for helping me resolve the problem.

closed 2020-09-29 by rakeshbci

cboulay, 2020-09-29:
@rakeshbci I guess that means that you had a rogue liblsl somewhere on your path. This might cause you trouble in the future.

Can you please unset the environment variable temporarily, then add a `print(libpath)` immediately before the [`lib = CDLL(libpath)` line](https://github.com/labstreaminglayer/liblsl-Python/blob/master/pylsl/pylsl.py#L1217) and try to import pylsl again?

If the printed path is different to the expected path then this was your problem and you should probably get this other library off the PATH. If it's the same dll then we still don't know what the issue was.

rakeshbci, 2020-09-29:
@cboulay It indeed was a problem due to rogue liblsl. I had installed an older version of muse lab which comes with a lsl.dll. When I used the print(libpath), I found the file in my muse folder. Thank you for the guidance.