labstreaminglayer/pylsl #34: PerformanceTest is incompatible with PyQtGraph 0.11

tstenner, 2020-09-18:
Almost all examples run without issues (after i `pip install pyqtgraph`) but `python PerformanceTest.py` fails with 
```
Created outlet with name BetaGen and type EEG
Created outlet with name GeneratedCentreOutMarkers and type Markers
looking for an EEG stream...
Reading from inlet named BetaGen with channels ['RAW1', 'SPK1', 'RAW2', 'SPK2', 'RAW3', 'SPK3'] sending data at 16384.0 Hz
Looking for stream with type Markers
Reading from inlet named GeneratedCentreOutMarkers with channels ['EventMarkers']
Traceback (most recent call last):
  File "PerformanceTest.py", line 308, in <module>
    qwindow.parent().setWindowTitle("pylsl PerformanceTest")
AttributeError: 'NoneType' object has no attribute 'setWindowTitle'
```

Edit:
But i get the same error in master (l300), so it must be sth else than your changes. `pyqtgraph==0.11.0`

_Originally posted by @agricolab in https://github.com/labstreaminglayer/liblsl-Python/pull/33#issuecomment-694803530_

agricolab, 2020-09-19:
PerformanceTest.py runs fine after pyqtgraph was uninstalled (with printouts), and with visualization after `pip install pyqtgraph==0.10`

agricolab, 2020-09-20:
```
if haspyqtgraph:
    qapp = pg.QtGui.QApplication(sys.argv)
    qwindow = pg.plot(title="pylsl PerformanceTest")
    qwindow.clear()
```
runs with pyqtgraph==0.11 and pyqtgraph==0.10

Could be linked to the deprecation of PlotWindow? 
https://pyqtgraph.readthedocs.io/en/latest/graphicswindow.html#pyqtgraph.PlotWindow