labstreaminglayer/pylsl #40: crash with sklearn on my system

xloem, 2021-01-06:
This happens only when pylsl is imported first:
```
>>> import pylsl
>>> import sklearn
*** Error in `python3': free(): invalid pointer: 0x00007f6446a63bc0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81299)[0x7f645fb39299]
/lib64/libstdc++.so.6(_ZNSt6locale5_Impl16_M_install_facetEPKNS_2idEPKNS_5facetE+0x142)[0x7f64467cded2]
/lib64/libstdc++.so.6(_ZNSt6locale5_ImplC1Em+0x1e3)[0x7f64467ce323]
/lib64/libstdc++.so.6(+0x71295)[0x7f64467cf295]
/lib64/libpthread.so.0(+0x620b)[0x7f646059520b]
/lib64/libstdc++.so.6(+0x712e1)[0x7f64467cf2e1]
/lib64/libstdc++.so.6(_ZNSt6localeC2Ev+0x13)[0x7f64467cf323]
/lib64/libstdc++.so.6(_ZNSt8ios_base4InitC2Ev+0xbc)[0x7f64467cc17c]
/home/user/.local/lib/python3.8/site-packages/scipy/fft/_pocketfft/pypocketfft.cpython-38-x86_64-linux-gnu.so(+0x77f0)[0x7f64464ae7f0]
/lib64/ld-linux-x86-64.so.2(+0xf9c3)[0x7f64610499c3]
/lib64/ld-linux-x86-64.so.2(+0x1459e)[0x7f646104e59e]
/lib64/ld-linux-x86-64.so.2(+0xf7d4)[0x7f64610497d4]
/lib64/ld-linux-x86-64.so.2(+0x13b8b)[0x7f646104db8b]
/lib64/libdl.so.2(+0xfab)[0x7f646038bfab]
/lib64/ld-linux-x86-64.so.2(+0xf7d4)[0x7f64610497d4]
/lib64/libdl.so.2(+0x15ad)[0x7f646038c5ad]
/lib64/libdl.so.2(dlopen+0x31)[0x7f646038c041]
/usr/local/lib/libpython3.8.so.1.0(_PyImport_FindSharedFuncptr+0x183)[0x7f6460c7d573]
/usr/local/lib/libpython3.8.so.1.0(_PyImport_LoadDynamicModuleWithSpec+0x2ae)[0x7f6460c344ae]
/usr/local/lib/libpython3.8.so.1.0(+0x24fb81)[0x7f6460c31b81]
/usr/local/lib/libpython3.8.so.1.0(+0x112e13)[0x7f6460af4e13]
/usr/local/lib/libpython3.8.so.1.0(PyVectorcall_Call+0x5d)[0x7f6460a961bd]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xa630)[0x7f6460a59860]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x13ab)[0x7f6460c02c4b]
/usr/local/lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x15f)[0x7f6460a9386f]                                                                                                      [85/118]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xa925)[0x7f6460a59b55]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x9023)[0x7f6460a58253]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x889a)[0x7f6460a57aca]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x889a)[0x7f6460a57aca]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x889a)[0x7f6460a57aca]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
/usr/local/lib/libpython3.8.so.1.0(+0xb22bc)[0x7f6460a942bc]
/usr/local/lib/libpython3.8.so.1.0(_PyObject_CallMethodIdObjArgs+0x10f)[0x7f6460a9452f]
/usr/local/lib/libpython3.8.so.1.0(PyImport_ImportModuleLevelObject+0x4ff)[0x7f6460c32f1f]
/usr/local/lib/libpython3.8.so.1.0(+0x21ad2f)[0x7f6460bfcd2f]
/usr/local/lib/libpython3.8.so.1.0(PyCFunction_Call+0x147)[0x7f6460a963b7]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xa630)[0x7f6460a59860]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x13ab)[0x7f6460c02c4b]
/usr/local/lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x15f)[0x7f6460a9386f]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x889a)[0x7f6460a57aca]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x13ab)[0x7f6460c02c4b]
/usr/local/lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x15f)[0x7f6460a9386f]
/usr/local/lib/libpython3.8.so.1.0(+0xb22bc)[0x7f6460a942bc]
/usr/local/lib/libpython3.8.so.1.0(_PyObject_CallMethodIdObjArgs+0x10f)[0x7f6460a9452f]
/usr/local/lib/libpython3.8.so.1.0(PyImport_ImportModuleLevelObject+0x62c)[0x7f6460c3304c]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xb500)[0x7f6460a5a730]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x13ab)[0x7f6460c02c4b]
/usr/local/lib/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x93)[0x7f6460c033c3]
/usr/local/lib/libpython3.8.so.1.0(PyEval_EvalCode+0x3b)[0x7f6460c0340b]
/usr/local/lib/libpython3.8.so.1.0(+0x21bbb2)[0x7f6460bfdbb2]
/usr/local/lib/libpython3.8.so.1.0(+0x112e13)[0x7f6460af4e13]
/usr/local/lib/libpython3.8.so.1.0(PyVectorcall_Call+0x5d)[0x7f6460a961bd]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xa630)[0x7f6460a59860]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x13ab)[0x7f6460c02c4b]                                                                                                   [52/118]
/usr/local/lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x15f)[0x7f6460a9386f]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xa925)[0x7f6460a59b55]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
/usr/local/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x9023)[0x7f6460a58253]
/usr/local/lib/libpython3.8.so.1.0(+0x6c0a1)[0x7f6460a4e0a1]
======= Memory map: ========
[... 53 lines cut ...]
7f6448159000-7f64481b9000 r-xp 00000000 fd:02 91067                      /usr/lib64/libpcre.so.1.2.0
7f64481b9000-7f64483b9000 ---p 00060000 fd:02 91067                      /usr/lib64/libpcre.so.1.2.0
7f64483b9000-7f64483ba000 r--p 00060000 fd:02 91067                      /usr/lib64/libpcre.so.1.2.0
7f64483ba000-7f64483bb000 rw-p 00061000 fd:02 91067                      /usr/lib64/libpcre.so.1.2.0
7f64483bb000-7f64483df000 r-xp 00000000 fd:02 91079                      /usr/lib64/libselinux.so.1
7f64483df000-7f64485de000 ---p 00024000 fd:02 91079                      /usr/lib64/libselinux.so.1
7f64485de000-7f64485df000 r--p 00023000 fd:02 91079                      /usr/lib64/libselinux.so.1
7f64485df000-7f64485e0000 rw-p 00024000 fd:02 91079                      /usr/lib64/libselinux.so.1
7f64485e0000-7f64485e2000 rw-p 00000000 00:00 0 
7f64485e2000-7f64485f8000 r-xp 00000000 fd:02 626528                     /usr/lib64/libresolv-2.17.so
7f64485f8000-7f64487f8000 ---p 00016000 fd:02 626528                     /usr/lib64/libresolv-2.17.so
7f64487f8000-7f64487f9000 r--p 00016000 fd:02 626528                     /usr/lib64/libresolv-2.17.so
7f64487f9000-7f64487fa000 rw-p 00017000 fd:02 626528                     /usr/lib64/libresolv-2.17.so
7f64487fa000-7f64487fc000 rw-p 00000000 00:00 0 
7f64487fc000-7f64487ff000 r-xp 00000000 fd:02 98056                      /usr/lib64/libkeyutils.so.1.5
7f64487ff000-7f64489fe000 ---p 00003000 fd:02 98056                      /usr/lib64/libkeyutils.so.1.5
7f64489fe000-7f64489ff000 r--p 00002000 fd:02 98056                      /usr/lib64/libkeyutils.so.1.5
7f64489ff000-7f6448a00000 rw-p 00003000 fd:02 98056                      /usr/lib64/libkeyutils.so.1.5
7f6448a00000-7f6448a0e000 r-xp 00000000 fd:02 91016                      /usr/lib64/libkrb5support.so.0.1
7f6448a0e000-7f6448c0e000 ---p 0000e000 fd:02 91016                      /usr/lib64/libkrb5support.so.0.1
7f6448c0e000-7f6448c0f000 r--p 0000e000 fd:02 91016                      /usr/lib64/libkrb5support.so.0.1
7f6448c0f000-7f6448c10000 rw-p 0000f000 fd:02 91016                      /usr/lib64/libkrb5support.so.0.1
7f6448c10000-7f6448c41000 r-xp 00000000 fd:02 91008                      /usr/lib64/libk5crypto.so.3.1
7f6448c41000-7f6448e40000 ---p 00031000 fd:02 91008                      /usr/lib64/libk5crypto.so.3.1Aborted (core dumped)
```

cboulay, 2021-01-06:
I see you're using python 3.8 but I need more detailed information.

Exactly what version of Python 3?
What version of scikit-learn and how was it installed?
How did you install pylsl? What version do you have?

Not that it's worth much, but I just did a quick test with python 3.8.5, scikit-learn 0.23.2, and the latest pylsl from github, **ON WINDOWS**, and I didn't encounter any problems. I don't have time to test in Ubuntu right now; maybe tomorrow.

cboulay, 2021-01-06:
I remembered I had a WSL2 Ubuntu 20.04 environment so I tested there quickly (this time pylsl from pypi) and also didn't have any problems.

xloem, 2021-01-06:
I had set the packages up with `pip3 install pylsl && pip3 install sklearn`.  I'm using a git checkout of python3.8.
```
$ python3 --version
Python 3.8.7+
cpython]$ git describe
v3.8.7-3-g285b38c3a8
```

It will take me some time to test any further.  I tried to install pylsl from git and I'm having some troubles setting it up; the setup isn't working right now anymore.

xloem, 2021-01-06:
Okay, it looks like I had some old liblsl.so laying around that was causing the issue.  Upgrading liblsl.so resolved it.  Thanks for your patience.

closed 2021-01-06 by xloem

xloem, 2021-01-07:
To clarify, the resolution to this bug is from installing pylsl from git, not from the binary liblsl.