labstreaminglayer/pylsl #41: Compilation error for macOS<10.15

sari-saba-sadiya, 2021-01-28:
Hi

I am using conda with python 3.6.12 on macOS 10.13.6 (high sierra). I successfully installed pylsl via pip but when running `import pylsl` I got the error below. This is a known macOS compatibility issue as pre 10.15 versions can not work with `____chkstk_darwin`.

I tried downloading and building the 1.13.1 release with the macOS10.13 precompiled code but I get an egg error that I was unable to solve (see attached file), I think there might be a setup.py file missing? Finally I downloaded the zip from github and tried to follow [the readthedocs guide](https://labstreaminglayer.readthedocs.io/dev/lib_dev.html) but it doesn't seem to work for version 1.13 which doesn't have a cmakefilelist or setup files.

Could you please point me to where there are high sierra compatible files, or a guide on how to build from the 1.13 release?

Thanks,
Sari
[egg_error.txt](https://github.com/labstreaminglayer/liblsl-Python/files/5890365/egg_error.txt)

> Traceback (most recent call last):
>   File "/Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/pylsl.py", line 1267, in <module>
>     lib = CDLL(libpath)
>   File "/Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/ctypes/__init__.py", line 348, in __init__
>     self._handle = _dlopen(self._name, mode)
> OSError: dlopen(/Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/lib/liblsl.dylib, 6): Symbol not found: ____chkstk_darwin
>   Referenced from: /Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/lib/liblsl.dylib (which was built for Mac OS X 10.15)
>   Expected in: /usr/lib/libSystem.B.dylib
>  in /Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/lib/liblsl.dylib
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/__init__.py", line 2, in <module>
>     from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
>   File "/Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/pylsl.py", line 1281, in <module>
>     raise RuntimeError(err_msg + "\n " + __dload_msg)
> RuntimeError: liblsl library '/Users/admin/opt/anaconda3/envs/psychopy/lib/python3.6/site-packages/pylsl/lib/liblsl.dylib' found but could not be loaded - possible platform/architecture mismatch.
>  
>  You can download the correct LSL library for your platform from the liblsl releases page assets: https://github.com/sccn/liblsl/releases

tstenner, 2021-01-29:
The pylsl release is mostly independent from the liblsl release, so you can use pylsl 1.14 with another liblsl version.
The easiest way to do this is with the `PYLSL_LIB` environment variable, e.g. `PYLSL_LIB=$PWD/liblsl64.so.1.13.1 python -m pylsl.examples.SendStringMarkers`.
You can get the newest liblsl binaries from the [release page](https://github.com/sccn/liblsl/releases) or [build it yourself](https://labstreaminglayer.readthedocs.io/dev/lib_dev.html).

I haven't got a Mac, but @cboulay know everything there is to know about OS X peculiarities.

sari-saba-sadiya, 2021-01-29:
Hi

@tstenner thank you for the prompt reply.

As I noted in my question, the binaries from the release page and building it myself (using the exact guide you linked to) both produce different errors due to what I believe might be missing files that are required by pip or cmake, namely a the cmake seems to be looking for a setup.py file or a cmakelist that are not provided.

Considering the error I linked in the egg_error.txt and the [discussion in this thread](https://github.com/pypa/pip/issues/5271#issue-315098549) it might have to do with mismatch between the #egg= and the project name in metadata. This happens when I simply `pip git+https://github.com/sccn/liblsl@1.13` so it is an issue that would effect many people.

Thanks

tstenner, 2021-01-29:
liblsl and liblsl-Python/pylsl are different projects. liblsl is a C++ library, and liblsl-Python/pylsl is a python wrapper that loads the liblsl binary. Your mac can't load the liblsl binary, so unless there's a simple fix I'm not aware of either you or someone else has to build liblsl. As I see you're using anaconda, you can try to let conda build it.

In any, case, you need to get a copy of the [liblsl repository](https://github.com/sccn/liblsl).
With conda-build, you can then run `conda build /path/to/liblsl/conda`, or you can install CMake and build it manually.

It's been a while since I have touched the python installer code, but from what I remember it produces `bdist_wheel`s, not eggs.

cboulay, 2021-01-29:
Hi @sari-saba-sadiya , it would be good to know where you went astray so we could update the documentation to be clearer.

As Tristan mentioned, liblsl and liblsl-Python are different things. But let me make it a bit clearer:
* `liblsl` is a compiled shared library. On a Mac this takes the form of a .dylib file, on windows it's a dll.
* liblsl-Python is an interface to `liblsl`, or what we might call a "wrapper". This takes the form of one-or-more python files collected in a Python package called `pylsl`.

Everything that uses labstreaminglayer in some way, including liblsl-Python (the project containing `pylsl`), needs a copy of the compiled liblsl library. For liblsl-Python, the compiled liblsl binary is packaged inside the wheel so when you do `pip install pylsl`, it also comes with the necessary file. We do our best to make sure the version of the dylib that ships with the packaged wheel is as compatible as possible, but, as you encountered, inevitably we encounter situations where it isn't compatible.

So the simple solution is to replace^1 the .dylib file that comes with pylsl with a .dylib that is compatible with your system. This doesn't have to be an old version of liblsl. In fact, we would prefer if you use the latest version available. You can try the [liblsl release page](https://github.com/sccn/liblsl/releases) to see if you can find a dylib that works on your system. If you can't find one then you can build it yourself. Again, this doesn't have to be an old version of liblsl, please use the most recent! If you are still having trouble building the most recent version of liblsl then let us know and I'll try to build one for you. But I'm running MacOS 11.1 so it might take me some time to figure out how to cross-compile for an older version.

^1 - I suggest that in the end you place the correct dylib file inside the pylsl package directory in your Python environment. Until then, while testing, follow Tristan's advice and use the `PYLSL_LIB` environment variable.

cboulay, 2021-01-29:
Try the attached file. I added `-DCMAKE_OSX_DEPLOYMENT_TARGET=10.13` to the cmake command. I have no idea if it took.

[liblsl-1.14.0-OSX_amd64.tar.bz2.zip](https://github.com/labstreaminglayer/liblsl-Python/files/5895996/liblsl-1.14.0-OSX_amd64.tar.bz2.zip)

sari-saba-sadiya, 2021-01-30:
Hi,

Thank you both for your help.

The liblsl1.14 did not work, namely I still got the `____chkstk_darwin` error. I had to restart with a clean conda env. I downloaded the source code for the liblsl 1.13 version, compiled with conda build (didn't work otherwise) and copied the dylib, this still did not work and I had to modify the pylsl file to make sure it found the dylib.

Now it all seems to be working as it should!

Regards,
Sari

closed 2021-01-30 by sari-saba-sadiya

cboulay, 2021-01-30:
> I downloaded the source code for the liblsl 1.13 version, compiled with conda build (didn't work otherwise)

What do you mean by this? Did you try to build with cmake or another method?

> The liblsl1.14 did not work
> the source code for the liblsl 1.13 version

Did you try the source for 1.14? Or just the precompiled, then the source for 1.13?

sari-saba-sadiya, 2021-01-31:
Hi,

I tried the source code you provided, previously I also tried the precompiled 1.14. I ended up compiling liblsl 1.13 with conda and it worked (yes previously I just tried cmake).