labstreaminglayer/pylsl #44: PyLSL in Docker

mlej8, 2021-07-03:
Hi, I'd like to run PyLSL to Docker, I was wondering which port of my container I should expose to get the data from the OpenBCI GUI ? Usually pylsl uses which port to establish the TCP connection ?

tstenner, 2021-07-03:
I'm afraid there's a bit more needed, unless you play around with the [lab.KnownPeers](https://labstreaminglayer.readthedocs.io/info/lslapicfg.html#lslapicfg-lab) configuration variable.
The [firewall configuration page](https://labstreaminglayer.readthedocs.io/info/network-connectivity.html#firewall-settings) lists the ports (as for the multicast groups, only one is needed). Last time I checked, Docker doesn't really play well with multicast, but it might have gotten better.

mlej8, 2021-07-03:
I am running pylsl inside of a container which exposed the following ports, "16571-16604:16571-16604", however I am unable to resolve_stream(), e.g. the application does not detect the stream of data whne I do "Start LSL Stream" from the OpenBCI GUI. Could you clarify if I should expose other additional ports ? Which port does the OpenBCI GUI uses when connecting the LSL stream ?

ddetommaso, 2021-07-04:
Hi @mlej8

did u try ti configure the container with network_mode: host and privileged runtime capabilities? With those two enabled you won't need to expose any ports because you'll share the host network resources and devices.

https://docs.docker.com/network/network-tutorial-host/

https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities

mlej8, 2021-07-04:
Hi @ddetommaso , thank you for your response, yes I tried running the container using the host's network with priviledged mode, but I still couldn't resolve_stream() using the OpenBCI_GUI Networking Test Kit LSL lslStreamTest.py. Have you tried and it worked?

ddetommaso, 2021-07-04:
Hi @mlej8 

I don't have a specific experience with OpenBCI but I have with integrating LSL in dockers. Using the aforementioned options you should be able to use the same network capabilities of your host machine, so maybe the issue it's of a different kind (GUI issue?). Enabling detailed debug info inside your application we can investigate more.

mlej8, 2021-07-04:
@ddetommaso, I am trying to run the following script: https://github.com/OpenBCI/OpenBCI_GUI/blob/master/Networking-Test-Kit/LSL/lslStreamTest.py inside of a docker container with `network=host` and `privileged` mode. When I do `start LSL stream` from the [OpenBCI Windows GUI](https://openbci.com/downloads) I don't receive the stream inside of my container, whereas when I run the same script on WSL, everything works as expected and I am able to get the stream to my terminal.

ddetommaso, 2021-07-04:
@mlej8  could u please detail more your experimental setup? Specifically, if the docker container it's running on Windows on the same machine of the server or on a different operating system. Did u try to access the low-level API of OpenBCI from the docker container?

mlej8, 2021-07-04:
The docker container is ran using Docker Desktop for Windows on WSL 2. The container is running on my machine (on WSL 2). The OpenBUI GUI is also opened on my machine. I have not tried the low-level API of OpenBCI, could you point me to the repo where I can find it ?

ddetommaso, 2021-07-05:
Hi @mlej8 

I don't have experience with WSL so I might be wrong but a few tests could make the situation clearer.
Running inside the same machine I suppose you have client and server on the same network. Did u test sending/receiving multicast packets between client and server? Specifically, you can verify if inside and outside the docker container you notice differences. If you're able to send/receive multicast packets then the issue is not network related.

The following command should tell you if multicast is enabled for your interface and the specific address
`netstat -g`

For testing if you are able to receive multicast packets you can combine the commands 
`ping <multicast address>` in one end 
and `tcpdump -i <network interface> -s 65534 host <multicast address>` in the other location.

tstenner, 2021-07-07:
> Could you clarify if I should expose other additional ports ?

It's not about the ports, it's about multicast packets. Basically, the outlet listens for stream resolution packets sent to a multicast group and replies with it's connection details to the sender. Docker doesn't receive these packets properly, so streams in a docker container can't be found.

mlej8, 2021-07-07:
But shouldn't this be solved by setting `network=host` for the Docker container? I tried that and was still not able to receive the stream inside of the container.

alastairhmoore, 2021-07-16:
@cboulay asked me to share my experience with a possibly-related issue
In my setup I have a stream generated inside WSL2 and I want to record it with LabRecorder on the host windows.  As mentioned in https://github.com/labstreaminglayer/liblsl-Python/issues/44#issuecomment-873459749 I was able to achieve this by specifying the IP address of the WSL instance (`hostname -I` from inside WSL) as a KnownPeer.
The other requirement was to allow LabRecorder through the firewall - both on private and public networks. This was done for the application rather than for any ports. I'm not using Docker at all but hope it helps anyway.

mdcg, 2024-07-18:
Hello! I have a similar problem. I'm exporting the data via LSL on my local computer and I'm trying to consume it in a script that is inside a Docker container. I already set `network_mode: host` and `privileged: true`, but it didn't work. I ran the script outside of Docker and the data was found without any problems. Has anyone ever experienced this? I'm sending some prints to illustrate what's going on:
[image]
[image]

dorianck, 2025-09-24:
@mdcg  I have the same issue than you. did you manage to solve it ?