labstreaminglayer/pylsl #45: Integration of Intel realsense with lsl

deolipankaj, 2021-08-09:
Hello everyone, 
I am working on the task where i have to synchronize the timestamps from 2 different sensors (i.e. Infrared images from Intel realsense and EEG signals (with open signals)). I have a separate script with which stream the images from intel realsense and I have the script (https://github.com/biosignalsplux/opensignals-samples/blob/master/LSL/Python/lsl_sample.py) which can give me the values at every timestamp. 
I have questions regarding how to integrate the realsense images with the lsl layer ?

cboulay, 2021-08-09:
Have you looked at the examples?
https://github.com/labstreaminglayer/liblsl-Python/blob/master/pylsl/examples/SendDataAdvanced.py

What are your questions?

dmedine, 2021-08-10:
I'm going to close this because it isn't really an issue that demands revision to the codebase. However, if you are available, I can recommend a free webinar about LSL programming next month that will probably answer a lot of your questions: http://www.diademics.com/#events.

closed 2021-08-10 by dmedine