labstreaminglayer/pylsl #46: Unable to stream data from multiple sources at the same time

tenghowe, 2021-09-21:
Hi,

I'm trying to stream from 2 different data sources (computers A and B) to another computer (C). Computers A and B are connected to C via 2 different network adapters.

When computers A or B are connected to C by themselves, there isn't any issue, but when both are connected to C, only the data stream from A or B will work, depending on the network adapter which has higher priority.

Has anyone faced similar issues before?

cboulay, 2021-09-21:
To make sure this gets the right visibility, we should probably make sure it's not specific to Python and then open an issue in sccn/liblsl.

Can you please go to the [liblsl releases page](https://github.com/sccn/liblsl/releases/tag/v1.15.2) and download both liblsl and the lslexamples package on each of your 3 computers.

Then on computer A ,run `SendStringMarkers`.
On computer B, run `SendDataSimple`
On computer C, open 2 different consoles, and in the first run `ReceiveStringMarkers`, and the next run `ReceiveDataSimple`.

If you reproduce the above problem then please close this issue and open a new one in the sccn/liblsl repo where it will be seen by more people and have higher priority.

If you cannot reproduce the problem, then maybe you need to take a look at your data sources to make sure their name/type/unique_id are not conflicting.

tenghowe, 2021-09-22:
I was able to reproduce the issues via the steps you mentioned. However, I managed to find a workaround by linking all 3 computers together via a switch.

Thanks for the help.

closed 2021-09-22 by tenghowe

cboulay, 2021-09-22:
Great, thanks for updating the issue.
For the record, I think this would probably be solved with https://github.com/sccn/liblsl/pull/31