labstreaminglayer/pylsl #47: LSL and PyInstaller

dsortinothimus, 2021-09-30:
Hi, I was trying to pack a small executable with PyInstaller on a small anaconda environment using few packages, pylsl among them. The script works fine if launched via terminal, but once packed in a .exe, I encounter this [image]. 

Have you ever experienced an issue of this kind?

cboulay, 2021-09-30:
I don't know enough about PyInstaller to help much. I can give some general information.

pylsl.py searches for `lsl.dll` in several locations. If you install pylsl with `pip install pylsl`, then it will download a wheel from pypi that includes this dll file that is then installed within the package folder. Alternatively, you can download `lsl.dll` from one of the [sccn/liblsl releases](https://github.com/sccn/liblsl/releases) and then place that in one of the search locations. The search locations are the standard windows ones, e.g. C:\Windows\System32, and include searching in a folder specified by an environment variable named `PYLSL_LIB`.

So my completely ignorant guess is that PyInstaller doesn't package lsl.dll unless you tell it to. You can either supply lsl.dll separately and put it in one of the search folders, or maybe there's a pyinstaller command switch that allows you to specify additional folders to glob binaries from. e.g., see [this SO answer](https://stackoverflow.com/a/38682416/1256069)

dsortinothimus, 2021-09-30:
Well, I wasn't so sure about the file that pylsl.py was searching, so I was pointing at the wrong lsl.dll

Just using the right path in PYLSL_LIB worked great for me. Thanks for your time!

cboulay, 2021-09-30:
Oh, looking a little closer at your error message, I see that pylsl thinks it's running from a temp folder. I guess the exe is expanding into a temp folder at runtime and that doesn't include the lib/lsl.dll , probably because it wasn't packaged into the exe originally.

Anyway if you're happy to use an environment variable then that's good enough. Please close the issue if you are satisfied.

closed 2021-09-30 by dsortinothimus

dmedine, 2021-09-30:
FWIW, I had some trouble with pyinstaller and LSL a while back, which I was able to solve. I also had trouble with logging module in the particular project I was working on which is probably of little interest here. I think the code that I had to modify in pylsl in order to get it to work has changed since then, but if more people are interested in this, it might be worth while to construct a short tutorial. 

Here are my notes from an email correspondence I had on the topic:

I figured out what the problem is with pyinstaller and liblsl. For some reason when using pyinstaller with scripts that call CDLL (which is what pylsl uses to load liblslxx.dll) CDLL cannot use full paths. The trick was to modify pylsl so that it stops looking for the path to the library. This was explained here: https://kezunlin.me/post/da9d93d6/.

Attached is the modified pylsl.py that I used. The lines of interest are around 1183-1201. The python interface has changed recently to reflect a choice to change the name of the LSL binaries or else I would point you to the source code online. 

There is a similar problem with the log file that I ran into. If you get an error like:
Traceback (most recent call last):
  File "main_GUI.py", line 655, in <module>
  File "main_GUI.py", line 409, in setup
  File "logging\config.py", line 71, in fileConfig
  File "logging\config.py", line 104, in _create_formatters
  File "configparser.py", line 958, in __getitem__
KeyError: 'formatters'
[2020] Failed to execute script main_GUI

It means that the logging module cannot find your logging.conf file which means that pyinstaller needs some extra love. Again, the problem has to do with full paths to resource files in the python source code. In this case, in Setup, main_GUI.py is looking for a full path rather than a filename on its own. So, I had to modify lines 409 of main_GUI.py (also attached). 

I am not sure what the answer is to this path problem. It seems to me that this is a flaw in the design of pyinstaller, but I could imagine that fixing it is very, very hard. For now, I don’t see any way around it other than the hack that I did. This means that for the future, we will have to maintain a special version of pylsl with that little patch I made just so it can get compiled properly by pyinstaller, which is not ideal. A more sophisticated solution would be a build script that automatically downloads the latest version, applies the patch to pylsl.py and then runs pyinstaller, but one thing at a time.

Here is the call to pyinstaller command that I used:

pyinstaller --onefile --paths=C:\Python37\Lib\site-packages --add-data=".\log\logging.conf;." --add-binary="C:\Python37\Lib\site-packages\pylsl\liblsl64.dll;lib" main_GUI.py

Note that after the semi-colon in the add data option, I specified the relative path to logging.conf relative to the generated executable. It could be that Mac will put this in the Resources folder or something like this, inside the .app/Contents bundle. You will have to check/research this. Obviously you will have to change the paths to the corresponding ones on your computer.

I also had to copy liblsl64.dll to the same folder as the main_GUI.exe file (i.e. the dist folder). Again, on OSX you can probably bundle this in Frameworks or something like that, but this will probably mean some modification of the .spec file.

Finally, I think on that on Mac you will have to change the semi-colon before ‘lib’ to a colon (found that info here: https://stackoverflow.com/questions/48210090/how-to-use-bundled-program-after-pyinstaller-add-binary but haven’t tested it).