labstreaminglayer/pylsl #49: Not receiving Data From Amplifier

ikirtivardhansingh, 2021-11-18:
Started ReceiveAndPlot.py but can't get any stream from the ANT EEGO.

agricolab, 2021-11-18:
How are you using to sending data via LSL from ANT eego? The sdk-app or their eego software? Can you verify it is actually sending?  What operating system are you using? Are you on the same machine as the Outlet? Is it an issue with the Outlet not being found or within ReceiveAndPlot.py?

It might be a network connectivity issue, see https://labstreaminglayer.readthedocs.io/info/network-connectivity.html

But we would need more information to answer this question better.

madhavgovindgmg, 2021-11-22:
> We were unable to installing the EEGO SDK on a Linux system so we were trying to directly read data from the device through the python code given in your repo.
> Currently we are using Ubuntu on a CPU powered by Intel i7 processor.
> Yes we are on the same machine as the outlet.

agricolab, 2021-11-22:
What makes you believe you could access the data from the device without the SDK and a corresponding streaming outlet? 

Afaik eemagine / Ant Neuro does not offer the SDK for Linux, and that means there are no drivers. Short of back-engineering, e.g. by listening to the traffic, or asking Ant Neuro whether they share the protocol with you, there isn't much i think one can do to run their amplifiers on linux. Probably easiest to switch to Windows. Didn't they sell you a Windows PC with the Amps? They usually do.

madhavgovindgmg, 2021-11-22:
Hey, Can I switch to Neuro electric??
I guess you would be the right person who can guide me here....
Does Neuro Electric have support for Linux??    
Also, Can I use it on a raspberry pi??

agricolab, 2021-11-22:
This discussion is already out of scope of this issue. Considering the issue was caused by a faulty setup with missing SDK,  i recommend you close it. 

If you need amplifiers that are supported on Linux, best you give manufactures a call, and ask them directly.

closed 2021-11-22 by agricolab