labstreaminglayer/pylsl #53: Byte order conversion on the raspberry pi.

vincentNRC, 2022-03-29:
Hi,

I've been experimenting with LSL on the Raspberry Pi in the past few days and so far so good, it's installed and I'm able to exchange some data with a Windows PC. 

The issue comes when I tried to send a string from the PC to the Pi. I keep getting an error from "data_receiver.cpp" line 342 that says "Stream transmission broke off (The byte order conversion requested by the other party is not supported.)".

I've had a look at the "data_receiver.cpp" script and I've identified that the error is thrown from line 231 when using the `can_convert_endian()` function. So I'm guessing this is returning false and the error is thrown.

Based on these observations, my questions is: Is there some conversion I'm not doing when sending strings from the PC to the Pi?

A good way to reproduce this is to run the "SendStringMarkers.py script" on the PC and the "ReceiveStringMarkers.py" on the Pi.

For reference, I'm using PyLSL version 1.15.0 on Raspberry Pi 3B.

Thanks in advance for your help.

Cheers,
Vincent G.

cboulay, 2022-03-29:
I need to put out a new pylsl release soon so I'll test this when I do that, but on a Pi 4B because that's what I have. I don't expect I'll be able to get to that until Friday at the earliest. Let me know if you make any more discoveries in the meantime.

MrMho, 2022-12-01:
Could you find a solution for this issue? I am getting the exact same error message when trying to send string markers from Neurobehavioral Systems' Presentation to PyLSL 1.16.0 (both running on the same Windows 10 machine).

tstenner, 2022-12-03:
See https://github.com/labstreaminglayer/App-LabRecorder/issues/87