labstreaminglayer/pylsl #56: pipenv install broken

hgrw, 2022-07-15:
On ubuntu 22, pipenv, version 2022.7.4

`$ pipenv install pylsl`

** pylsl binary is installed as a file called 'lib' **
`$ ls $HOME/.local/share/virtualenvs/mico-rVa5xFn2/lib/python3.10/site-packages/pylsl`

``` >>> import pylsl
...
RuntimeError: LSL binary library file was not found. Please make sure that the binary file can be found in the package lib folder
```

** Temporary fix **
`$ mv $HOME/.local/share/virtualenvs/mico-rVa5xFn2/lib/python3.10/site-packages/pylsl/lib $HOME/.local/share/virtualenvs/mico-rVa5xFn2/lib/python3.10/site-packages/pylsl/liblsl.so`

cboulay, 2022-07-15:
> pip installer creates binary file called 'lib'

Can you elaborate on this? That shouldn't be happening. The pip installer on non-Windows platforms should not have any binaries. The liblsl binaries are supposed to be retrieved separately.

What version of pylsl is this installing?

hgrw, 2022-07-17:
With a fresh (pipenv) environment, I can see that no binary is installed. Nonetheless, I still get the same runtime error RuntimeError: LSL binary library file was not found. It's looking in `...site-packages/pylsl/lib`. If no binary is supposed to be installed, why is it looking for one?

pylsl version: 1.16.0

cboulay, 2022-07-17:
The Windows and Mac wheels ship with a binary. At one point there was a Linux wheel too and that shipped with a binary, but it became too difficult to keep up with the `manylinux` requirements. Even though we were able to meet the requirements of the manylinux checkers, there were some distros that were incompatible and the presence of the binary actually made it much harder for those users.

So now, in Linux at least, we don't ship a binary and expect the user to install liblsl separately.
https://github.com/sccn/liblsl/releases

The error message that you're seeing should be much longer:

`LSL binary library file was not found.  Please make sure that the binary file can be found in the package lib folder or the system search path. Alternatively, specify the PYLSL_LIB environment variable.`

Focus on the system search path or PYLSL_LIB environment variable part, not the package lib folder.

hgrw, 2022-07-17:
Ah right - thanks! It would be helpful if this was documented either in the docs or readme.

closed 2022-07-17 by hgrw

cboulay, 2022-07-18:
I added some info to the README. Please let me know whether you think this is sufficient.

https://github.com/labstreaminglayer/liblsl-Python/blob/master/README.md#liblsl-dependency

hgrw, 2022-07-20:
Yep, no problem - thanks