labstreaminglayer/pylsl #59: Floating point precision

NielsonJ, 2022-11-10:
Hi,

I'm trying to send floating numbers over LSL but on the receiving end the data is quite mangled.

I tried both `cf_float32` and `cf_double64` dtypes. 

```python
from pylsl import StreamInfo, cf_float32, cf_double64

stream = StreamInfo(name="send_test_loop", type="command", channel_count=2, nominal_srate=0, channel_format=cf_float32, source_id="send_test")
outlet = stream_outlet(stream)

while True:
    outlet.push_sample([1_667_829_964.9174, 1_667_829_879.5164])
```

I have captured the output with [LabRecorder](https://github.com/labstreaminglayer/App-LabRecorder), saved the result as an  `.xdf` file and inspected it.

The resulting values if using `cf_float32` in the `.xdf` files are:
```python
array([1.6678300e+09, 1.6678299e+09], dtype=float32)

[0:2] : [1667830000.0, 1667829900.0]
dtype: dtype('float32')
max: 1667830000.0
min: 1667829900.0
shape: (2,)
size: 2
```

The resulting values if using `cf_float64` in the `.xdf` files are:
```python
array([1.66782996e+09, 1.66782988e+09])

[0:2] : [1667829964.9174, 1667829879.5164]
dtype: dtype('float64')
max: 1667829964.9174
min: 1667829879.5164
shape: (2,)
size: 2
```

Neither of which are the expected (IEEE 754) precision's. `pylsl.py` states the following:
```python
# For up to 24-bit precision measurements in the appropriate physical unit (
# e.g., microvolts). Integers from -16777216 to 16777216 are represented
# accurately.
cf_float32 = 1
# For universal numeric data as long as permitted by network and disk budget.
#  The largest representable integer is 53-bit.
cf_double64 = 2
```
But even if `cf_float32` is encoded using a 24-bit (binary?) floating point encoding I would expect a better precision. And from a 64-bit binary floating point encoding I would also expect an higher precision then is achieved. 

What is going on here, am I missing something?

agricolab, 2022-11-13:
```
1_667_829_964.917_448_951_321
```
returns `1667829964.917449`, so you are already losing precision during instantiation.
```
ctypes.c_float(1667829964.917449)
```
returns `c_float(1667830016.0)`

so alltogether, you are already sending floats with reduced resolution.

agricolab, 2022-11-13:
My solution would be to send strings, probably to encode `decimal.Decimal`, because jsons would also suffer from Pythons limited precision, like 
```decimal.Decimal("1_667_829_964.917_448_951_321")```

agricolab, 2022-11-13:
To be honest, i can't wrap my mind why you would receive`1667830000`, though. According to https://www.h-schmidt.net/FloatConverter/IEEE754.html, only `1667830016.0` and `1667829888.0` can be properly encoded.

NielsonJ, 2022-11-14:
Hi @agricolab, thank you for quick your response. 

You are right, I misunderstood the precision of single- and double-precision floating point implementations. Indeed, encoding `1_667_829_964.917_448_951_321` should yield `1667830016` in float32/binary32.

I also found out why I  received `1667830000` instead of `1667830016`, or actually seemed so. The change in value was due to the fact I didn't specify how the number should be printed, and this in turn changed the perceived values, I actually did receive `1667830016`.

For example, if you use `ctypes` the printed value is as expected. 
```python
ctypes.c_float(1_667_829_964.917_448_951_321)
```
returns `c_float(1667830016.0)` 

However, if you use `numpy` on the other hand.
```python
numpy.float32(1_667_829_964.917_448_951_321)
```
it returns/prints `1667830000.0`

If we on the other hand specify how to print.
```python
"%.3f" % numpy.float32(1_667_829_964.917_448_951_321)
```
it prints the expected value: `'1667830016.000'

My debugger showed me all this time the value of `1667830000.0`. 

Thank you for your time, and we will indeed find a solution using strings.

closed 2022-11-14 by NielsonJ

agricolab, 2022-11-15:
Thanks for the detective work, that's a really interesting issue with the numpy floats.

Good luck with the project!