labstreaminglayer/pylsl #60: standalone-compilation not working on Raspberry Pi

nicho98, 2022-11-29:
i triend to execute the standalone_compilation_linux.sh on my raspberry pi 3 after cloning the entire sccn/liblsl file into my liblsl-Python/pylsl/lib. However, once i execute the standalone_compilation_linux.sh, a list of error appear. 

*/thirdparty/loguru/loguru.cpp:907:91: error: invalid conversion from 'int' to 'loguru::close_handler_t {aka void (*)(void*)}' [-fpermissive]
thirdparty/loguru/loguru.cpp:907:91: error: invalid conversion from 'loguru::flush_handler_t {aka void (*)(void*)}' to 'unsigned int' [-fpermissive]
thirdparty/loguru/loguru.cpp: At global scope:
thirdparty/loguru/loguru.cpp:913:33: error: redefinition of 'const char* loguru::get_verbosity_name'
  const char* get_verbosity_name(Verbosity verbosity)
                                 ^
In file included from thirdparty/loguru/loguru.cpp:23:0:
thirdparty/loguru/loguru.hpp:593:14: error: 'const char* loguru::get_verbosity_name' previously defined here
  const char* get_verbosity_name(Verbosity verbosity);
              ^
*/ and many other error

tstenner, 2022-12-04:
What compiler do you have installed? (`gcc --version`)?

Depending on your environment, some flags might cause problems. Could you run the script with `bash -e standalone_compilation_linux.sh` and paste the full output?

cboulay, 2023-04-17:
Hi @TechStranger . Another user was having trouble in #68 but with some discussion they got things working. Given the success there, and lack of updates here, I'm closing this issue.

closed 2023-04-17 by cboulay