labstreaminglayer/pylsl #72: What data type can we use for channel_format?

anbara, 2023-10-04:
Hello, 

When I created a python script that stream a sensor data, I had a problem.
I can't find out the data type we can use while I looked forward to any documents.
(I think the information of data type we can should be written on [liblsl doc](https://labstreaminglayer.readthedocs.io/projects/liblsl/ref/streaminfo.html))

Finally, I got this information from `pylsl/pylsl.py`.
Maybe we can use `undefined`, `float32`, `double64`, `string`, `int32`,  `int16`, `int8`,  and `int64`, right?

I want someone add to the data type info to the document web site.

Thank you :)

cboulay, 2023-10-04:
Here are the formats for liblsl:
https://github.com/sccn/liblsl/blob/45017108067125c223011f994536bf50914afbc1/include/lsl_cpp.h#L57-L82

And they are available on the website:
https://labstreaminglayer.readthedocs.io/projects/liblsl/ref/enums.html#_CPPv420lsl_channel_format_t

For pylsl, there is no documentation website. You have to look at the examples and source. I've updated the SendDataAdvanced.py example with some additional details:
https://github.com/labstreaminglayer/pylsl/blob/13853cf17e7e3622c3984356308402956f310d40/pylsl/examples/SendDataAdvanced.py#L30-L34

anbara, 2023-10-10:
Thank you for adding the details on example code! I hope the new LSL users will not have the same problems as me.

closed 2023-10-10 by anbara