labstreaminglayer/pylsl #76: LSL binary library file was not found on raspberry pi

Lim0304, 2023-12-02:
I also have the issue：
LSL binary library file was not found. Please make sure that the binary file can be found in the package lib folder (/home/boris/.local/lib/python3.9/site-packages/pylsl/lib) or the system search path.
But when I try the method in other answer ：bash -e standalone_compilation_linux.sh, it have some problems , and didn’t generate liblsl.so file. How can I solve this？
the output paste:
pi@raspberrypi:~ $ cd liblsl
pi@raspberrypi:~/liblsl $ bash -e standalone_compilation_linux.sh
++ git describe --tags HEAD
+ echo v1.16.2-22-g4210bb6e
v1.16.2-22-g4210bb6e
+ g++ -fPIC -fvisibility=hidden -O2 -Ilslboost -DBOOST_ALL_NO_LIB -DASIO_NO_DEPRECATED -DLOGURU_DEBUG_LOGGING=0 '-DLSL_LIBRARY_INFO_STR="v1.16.2-22-g4210bb6e"' src/api_config.cpp src/buildinfo.cpp src/cancellation.cpp src/common.cpp src/consumer_queue.cpp src/data_receiver.cpp src/info_receiver.cpp src/inlet_connection.cpp src/lsl_inlet_c.cpp src/lsl_outlet_c.cpp src/lsl_resolver_c.cpp src/lsl_streaminfo_c.cpp src/lsl_xml_element_c.cpp src/netinterfaces.cpp src/resolve_attempt_udp.cpp src/resolver_impl.cpp src/sample.cpp src/send_buffer.cpp src/socket_utils.cpp src/stream_info_impl.cpp src/stream_outlet_impl.cpp src/tcp_server.cpp src/time_postprocessor.cpp src/time_receiver.cpp src/udp_server.cpp src/util/cast.cpp src/util/endian.cpp src/util/inireader.cpp src/util/strfuns.cpp thirdparty/pugixml/pugixml.cpp -Ithirdparty/pugixml thirdparty/loguru/loguru.cpp -Ithirdparty/loguru -Ithirdparty/asio -shared -o liblsl.so -lpthread -lrt -ldl
In file included from /usr/include/c++/8/vector:69,
                 from src/netinterfaces.h:5,
                 from src/resolve_attempt_udp.h:6,
                 from src/resolve_attempt_udp.cpp:1:
/usr/include/c++/8/bits/vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry&}; _Tp = asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry; _Alloc = std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry>]':
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type 'std::vector<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry, std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry> >::iterator' {aka '__gnu_cxx::__normal_iterator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry*, std::vector<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry, std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry> > >'} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:64,
                 from src/netinterfaces.h:5,
                 from src/resolve_attempt_udp.h:6,
                 from src/resolve_attempt_udp.cpp:1:
/usr/include/c++/8/bits/stl_vector.h: In member function 'void asio::detail::epoll_reactor::schedule_timer(asio::detail::timer_queue<Time_Traits>&, const typename Time_Traits::time_type&, typename asio::detail::timer_queue<Time_Traits>::per_timer_data&, asio::detail::wait_op*) [with Time_Traits = asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> >]':
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry*, std::vector<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry, std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry> > >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:69,
                 from src/common.h:12,
                 from src/resolver_impl.h:5,
                 from src/resolver_impl.cpp:1:
/usr/include/c++/8/bits/vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const lsl::stream_info_impl&}; _Tp = lsl::stream_info_impl; _Alloc = std::allocator<lsl::stream_info_impl>]':
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type 'std::vector<lsl::stream_info_impl>::iterator' {aka '__gnu_cxx::__normal_iterator<lsl::stream_info_impl*, std::vector<lsl::stream_info_impl> >'} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:64,
                 from src/common.h:12,
                 from src/resolver_impl.h:5,
                 from src/resolver_impl.cpp:1:
/usr/include/c++/8/bits/stl_vector.h: In member function 'std::vector<lsl::stream_info_impl> lsl::resolver_impl::results(uint32_t)':
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<lsl::stream_info_impl*, std::vector<lsl::stream_info_impl> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:69,
                 from src/common.h:12,
                 from src/resolver_impl.h:5,
                 from src/resolver_impl.cpp:1:
/usr/include/c++/8/bits/vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry&}; _Tp = asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry; _Alloc = std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry>]':
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type 'std::vector<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry, std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry> >::iterator' {aka '__gnu_cxx::__normal_iterator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry*, std::vector<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry, std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry> > >'} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:64,
                 from src/common.h:12,
                 from src/resolver_impl.h:5,
                 from src/resolver_impl.cpp:1:
/usr/include/c++/8/bits/stl_vector.h: In member function 'void asio::detail::epoll_reactor::schedule_timer(asio::detail::timer_queue<Time_Traits>&, const typename Time_Traits::time_type&, typename asio::detail::timer_queue<Time_Traits>::per_timer [... cut ...] y_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry*, std::vector<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry, std::allocator<asio::detail::timer_queue<asio::detail::chrono_time_traits<std::chrono::_V2::steady_clock, asio::wait_traits<std::chrono::_V2::steady_clock> > >::heap_entry> > >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~

Lim0304, 2023-12-02:
Waiting your reply.....

Lim0304, 2023-12-02:
My gcc versions is 8.3.0. Should I change the version of it？

tstenner, 2023-12-02:
I pushed an commit that adds an explicit cast. This should work with GCC 8.3.

Lim0304, 2023-12-02:
Can I repeat the previous operation now?

tstenner, 2023-12-03:
Sure. I'm couldn't test it with my GCC 12, but clang compiles it in the non-permissive mode.

Lim0304, 2023-12-04:
Thank you！it works.

cboulay, 2023-12-11:
FYI I added a liblsl deb file for bookworm (latest Rapsberry Pi OS) here; https://github.com/sccn/liblsl/releases/tag/v1.16.2