labstreaminglayer/pylsl #77: How to get the channel name of streamer and the length of channel is different.

PeiyangLin, 2023-12-18:
Hi! I'm using pylsl tool to grab the EEG signal to do some realtime data processing and encountered some difficulties.
1, The device is a 128ch eeg device with 8 extra bipolar electrodes, but when i use inlet.pull_sample(), the length of the list is only 90.
2, I want to know how to get the channel name when I initialize the streamer so I can pick the right channel?
I am looking forward to your reply, which will be of great help to me, thank you! :)

cboulay, 2023-12-18:
What is the outlet that is sending the EEG data? Is it a custom application, an open source application, or does it come from a manufacturer-provided LSL integration?

PeiyangLin, 2023-12-18:
Manufacturer-provided LSL integration, i turned it on in the software.
[image]

cboulay, 2023-12-18:
OK, well then you'll have to ask the manufacturer why there are only 90 channels in the stream.

As for finding the channel names, you need to parse the stream info which you get from the inlet.
[Here's an example](https://github.com/intheon/stream_viewer/blob/master/stream_viewer/data/stream_lsl.py#L156-L174) of how that can be done.

I'm closing this issue for tracking purposes because there's nothing to be done on our end to help. You may continue to ask questions here and I will be notified about them.

closed 2023-12-18 by cboulay

cboulay, 2023-12-18:
P.S., before you go too deep into creating your own real time processing framework, be sure to check out the already-existing BCI frameworks: NeuroPype, ezmsg, BRAND, OpenViBE, BCI2000, TimeFlux. I'm sure there's an MNE-based version too but I get a bit confused in the MNE ecosystem.