labstreaminglayer/pylsl #81: Not working with wireless connection

alexfr98, 2024-05-22:
Hello!

First, thank you for your awesome package. I was using it with the Oculus Quest 2, cable link, and my computer and it works perfectly fine. 
Now I want to do it on a different computer. With one, I will have the Unity application which will be the one sending a marker, and on another computer, I have the Python string marker receiver code, which has to receive the information. This setup is not working when these two codes are on different computers connected to the same private network. The tests were run in Windows, MacOS and with two different private networks.

I already set up the firewall settings to accept all types of network communication (private and public) in Unity and Python. 

Do you know if the package should work in a wireless connection? I understand that LSL uses UDP Broadcasting, so I think it should work. but I'm not sure if I did all the required steps.  
The Python  code is the same that in you example ReceiveStringMarkers example: https://github.com/labstreaminglayer/pylsl/blob/master/pylsl/examples/ReceiveStringMarkers.py

Code in Unity (really similar to your sample):

```
public class LSLOutletManager : MonoBehaviour
{
    public static LSLOutletManager Instance { get; private set; }
    /*
    * If you are instead trying to log a stimulus event then there are better options. Please see the 
    * LSL4Unity SimpleStimulusEvent Sample for such a design.
    */
    string StreamName = "LSL4Unity.ContentModerator";
    string StreamType = "Markers";
    private StreamOutlet outlet;
    private string[] sample = { "" };
    private void Awake()
    {
        // If there is an instance, and it's not me, delete myself.

        if (Instance != null && Instance != this)
        {
            Destroy(this);
        }
        else
        {
            Instance = this;
        }
    }

    void Start()
    {
        var hash = new Hash128();
        hash.Append(StreamName);
        hash.Append(StreamType);
        hash.Append(gameObject.GetInstanceID());
        StreamInfo streamInfo = new StreamInfo(StreamName, StreamType, 1, LSL.LSL.IRREGULAR_RATE,
            channel_format_t.cf_string, hash.ToString());
        outlet = new StreamOutlet(streamInfo);
        PushToOutlet("1");
    }

    public void PushToOutlet(string message)
    {
        if (outlet != null)
        {
            Debug.Log(message);
            sample[0] = message;
                
            // Debug.Log(sample[0]);
            outlet.push_sample(sample);
        }
    }

    private void OnApplicationQuit()
    {
        outlet.Close();
    }

}

```

Thank you very much and sorry to disturb you! Hopefully, we will be able to solve this!

Greetings,
AF

alexfr98, 2024-05-22:
I tried also the wireless connection using only Python and it works fine! I was using the examples SendStringMarkers.py and ReceiveStringMarkers.py

cboulay, 2024-05-22:
Hi @alexfr98 , I'm going to close this because it seems like a Unity-specific issue, and not confined to pylsl.

Actually, it's probably the most common complaint recently that LSL does not work properly across multiple devices when one of them is in Unity. If you join the [support Slack](https://join.slack.com/t/labstreaminglayer/shared_invite/enQtMzA2NjEwNDk0NjA5LTcyYWI4ZDk5OTY5MGI2YWYxNmViNjhkYWRhZTkwYWM0ODY0Y2M0YzdlZDRkZTg1OTUwZDU2M2UwNDgwYzUzNDg) then you'll see already some conversations about LSL in Unity there. 

From what I can tell, one or several of the following things are happening:

* Unity creates a network shim in Editor mode. Sometimes this shim ends up in a bad state and needs to be reset. Toggling the system's network adapter settings seems to help.
* The shim is bad at UDP multicast, which is what LSL uses for stream discovery. UDP multicast can be bypassed through using lsl_api.cfg and KnownPeers parameter.
* Unity is more likely to hold on to a zombie LSL thread if you have the LSL-containing object selected when you stop the editor. You need to select a different non-LSL object and start-stop to clear this thread.

None of this is too definitive, just based on what others have said.

closed 2024-05-22 by cboulay