labstreaminglayer/pylsl #84: int32 value type incorrect on Windows

matthiasdold, 2024-07-17:
I encountered an incorrect value type when using `int32` values on Window (tested with python 3.11 and 3.12, pip 24.0, pylsl 1.16.1). The minimum example below correctly prints `<class 'ctypes.c_int'>` both on Linux (current Arch) and MacOS (Darwin, M1). 

But on Window 11 it prints `<class 'ctypes.c_long'>` which would be int64 data type. Also the data transfer fails if you pull the data into an numpy.int32 buffer.

Minimum example:
```python
import pylsl

feature_info = pylsl.StreamInfo(
    name="test",
    type="EEG",
    channel_count=1,
    nominal_srate=1,
    channel_format=pylsl.cf_int32,
    source_id="testid",
)
outlet = pylsl.StreamOutlet(feature_info)

streams = pylsl.resolve_byprop("name", "test", timeout=1)
inlet = pylsl.StreamInlet(streams[0])

print(inlet.value_type)
```

I also checked the other data types from `pylsl.string2fmt` on all three OSs and all others works correctly.

cboulay, 2024-12-10:
Hi @matthiasdold, I added a unit test and it doesn't seem to trigger this error. Can you confirm that the unit test is testing the write thing?
https://github.com/labstreaminglayer/pylsl/blob/main/test/test_format.py

matthiasdold, 2024-12-12:
Hi @cboulay , 
thanks for looking into this!

I could not replicate the issue with your test, so I dug deeper. I now understand that this is not a `pylsl` problem, but an unexpected behavior. Therefore, here in detail, as it might help others in the future:

It all boils down to `ctypes.c_long` being different data types on Windows and Unix systems - [32bit for Windows and 64bit on Unix](https://stackoverflow.com/questions/3323778/is-pythons-ctypes-c-long-64-bit-on-64-bit-systems).

The problem I encountered can be replicated by this example:

```python
import ctypes
import time
import pylsl
import numpy as np

def test_with_derived_buffer_data_type():

    # Create outlet -> inlet pair
    feature_info = pylsl.StreamInfo(
        name="test",
        type="EEG",
        channel_count=1,
        nominal_srate=1,
        channel_format=pylsl.cf_int32,
        source_id='testid123',
    )
    outlet = pylsl.StreamOutlet(feature_info)
    streams = pylsl.resolve_byprop("name", "test", timeout=1)
    inlet = pylsl.StreamInlet(streams[0])

    # Derive the buffer from the value_type (incorrectly assuming c_long is int64 on all OS!)
    dtype_map = {
        ctypes.c_int: np.int32,
        ctypes.c_long: np.int64,
    }

    # Buffer will be int64, but data is int32
    n_buffer = 10_000                           # this needs to be large enough
    buffer = np.zeros(n_buffer).astype(dtype_map[inlet.value_type])
    data = np.ones(5).astype(np.int32)          # as the outlet was specified for int32

    n_push = 2             
    times = []

    # try a few times to get the data
    i = 0
    while i < 10 and len(times) < 1:
        print(times)
        outlet.push_chunk(data[:n_push])            # we need to push more than 1 for it to fail! (oddly, once we pushed more than 1 sample, the next time we just push 1, it also fails.)
        time.sleep(0.5)
        _, times = inlet.pull_chunk(dest_obj=buffer)
        i += 1

    assert (buffer[:n_push] == data[:n_push]).all(), f"{data[:n_push]=} and {buffer[:n_push]=} do not match!"
```

The problem therefore is by assuming the `dtype_map` between `ctypes` and numpy types, as defined above, would work on all OS, which it does not!

One should therefore refrain from comparing the `inlet.value_type` to `ctypes` for deriving the streams data type and instead __always use the `inlet.info().channel_format()`__ and match it against the `pylsl.cf_*` types (enum values).

closed 2024-12-12 by matthiasdold

agricolab, 2024-12-12:
The only thing you can rely on is that `c_long` is at least 32 bytes, meaning that this is not limited to the listed OS, but a general problem. See also https://en.cppreference.com/w/cpp/language/types

agricolab, 2024-12-13:
Numpy writes in regard to the data model for generic int and long dtypes, claiming that `numpy.int` is  *64bit on 64bit systems and 32bit on 32bit systems* and `long` is an alias of numpy.int* 

But numpy offers several C compatible dtypes, e.g.  `numpy.intc` which is compatible with C int and `numpy.uintc` which is compatible with C uint. 

For intc, it mentions that the alias is platform specific, e.g 
>Canonical name:
>  `numpy.intc`
> Alias on this platform (Linux x86_64):
> `numpy.int32`: 32-bit signed integer (-2_147_483_648 to 2_147_483_647).

https://numpy.org/doc/2.1/reference/arrays.scalars.html#numpy.intc

Maybe you want to give those C-compatible dtypes (i.e. `numpy.intc` and `numpy.longlong` a try and see how they behave?  (although we should probably also test whether this stil works when the datamodel for outlet and inlet differs)

matthiasdold, 2024-12-13:
Thanks for the detail @agricolab!

I will have a look over the weekend and come up with a test in case it fails. For my application (which creates a ring buffer with data type derived from the inlet, I get correct behavior when using `inlet.info().channel_format()` (https://github.com/matthiasdold/dareplane-pyutils/blob/f83f38647f3a871021847fb9de7eef92b86a27af/src/dareplane_utils/stream_watcher/lsl_stream_watcher.py#L164C1-L174C78)