labstreaminglayer/pylsl #85: Ping-pong first read sample is always empty

aleksandrskoselevs, 2024-07-26:
Please see the minimal reproducible example here 
https://github.com/aleksandrskoselevs/pylsl_ping_pong

I have 2 small scripts, `ping.py` and `pong.py`

1. Ping sends a number to Pong. 
2. Pong sends the same number back to Ping. 
3. Ping then increases that number by 1. And it begins again.

This is managed through 2 streams. 

**Expected output**

`ping.py`
```
Ping stream created.
Looking for a Pong stream...
Pong stream found.

Sent to Pong: [1]
Received from Pong: [[1]], at [1210523.46006125]
Sent to Pong: [2]
Received from Pong: [[2]], at [1210524.46006125]
Sent to Pong: [3]
Received from Pong: [[3]], at [1210535.508698666]
...
```

`pong.py`
```
Pong stream created.
Looking for a Ping stream...
Ping stream found.

Received from Ping: [[1]] at [1210512.850981666]
Sent to Ping: [[1]]
Received from Ping: [[2]] at [1210523.872085166]
Sent to Ping: [[2]]
Received from Ping: [[3]] at [1210534.882665333]
Sent to Ping: [[3]]
```

**Actual output**

`ping.py`
```
Ping stream created.
Looking for a Pong stream...
Pong stream found.

Sent to Pong: [1]
Received from Pong: [], at [] ### <-- empty first sample
Sent to Pong: [2]
Received from Pong: [[2]], at [1210524.46006125]
Sent to Pong: [3]
Received from Pong: [[3]], at [1210535.508698666]
...
```

`pong.py`
```
Pong stream created.
Looking for a Ping stream...
Ping stream found.

Received from Ping: [[1]] at [1210512.850981666]
Sent to Ping: [[1]]
Received from Ping: [[2]] at [1210523.872085166]
Sent to Ping: [[2]]
Received from Ping: [[3]] at [1210534.882665333]
Sent to Ping: [[3]]
```

As you can see, the first sample read by Ping, after being returned by Pong, is always empty.

Any insight is greatly appreciated. Thank you.

dmedine, 2024-07-31:
See what happens when you get rid of you 5 second delay after the first push out of ping. I think what is happening is that during the sleep ping misses the marker sent back from pong and this is what causes the [] as the first sample after the wait. If you drop the wait you will still get a [] as the first sample in ping (it pulls before ping receives and pushes back), but then you will get a [[1],[2]] the next time around in pong (and ping) before it then settles into the expected regime.

I am not sure how long the lifespan of a marker on the network is, but I guess it looks like less than 5 seconds.

Here is my output of ping.py (5 second sleep removed):
```
Ping stream created.
Looking for a Pong stream...
Pong stream found.
Sent to Pong: [1]
Received from Pong: [] at []
Sent to Pong: [2]
Received from Pong: [[1], [2]] at [1729722.2033211, 1729722.2033211]
Sent to Pong: [3]
Received from Pong: [[3]] at [1729723.2056324]
```

And pong (unchanged):
```
Pong stream created.
Looking for a Ping stream...
Ping stream found.
Received from Ping: [[1], [2]] at [1729721.9817811, 1729721.98579]
Sent to Ping: [[1], [2]]
Received from Ping: [[3]] at [1729722.9880421]
Sent to Ping: [[3]]
```

aleksandrskoselevs, 2024-07-31:
Thank you very much @dmedine. This, combined with your other advice on adding a length check resolved the issue. We are now getting the expected result.

I'm closing the Issue, but for completeness would like to know what the actual lifespan of a marker is. Our expectation was that it remains buffered indefinitely, until read. Should anyone know, please comment, since it is not clear from the docs.

closed 2024-07-31 by aleksandrskoselevs

dmedine, 2024-08-01:
Now that I have some more time to think about it, it doesn't make sense to me that a marker can disappear. It should be just like any other sample. The sample rate is only used for meta data and shouldn't affect behavior otherwise. That means that an inlet with a sampling rate of 0 should also buffer until a call to pull grabs it. The marker will be quite delayed, but its timestamp should reflect when it was originally sent (on the outlet host). I honestly don't know why this is happening.