labstreaminglayer/pylsl #88: Trouble catching LostError?

dmedine, 2024-11-18:
I am trying to do a pretty basic detection if a stream connection is lost (sorry, crappy formatting):

           ```
 try:
        s, _ = self.inlet.pull_sample(0)
       # do stuff
    except LostError:
        self.is_connected = False
```

but if I put a break point on the exception handler and destroy the attached outlet, I don't break on that line. I do get an error message that bubbles up from the C++ and prints out on my python debugger console:

`2024-11-18 21:42:12.778 (   4.343s) [R_lr_probabili  ]      data_receiver.cpp:344    ERR| Stream transmission broke off (Input stream error.); re-connecting...`

but I would expect that the path this should take is to return an error code of -2 that would then get handled by the python wrapper. Am I doing this wrong somehow? It seems very unlikely to me that such a bug would ever have gone unnoticed.

closed 2024-11-18 by dmedine

cboulay, 2024-11-18:
Somebody was asking about this recently. Did you have a resolution?

dmedine, 2024-11-18:
Yeah. It doesn't throw until the timeout expires. You have to set the `recover` flag to false if you want a `LostError` straight away on disconnect. (https://github.com/labstreaminglayer/pylsl/blob/21fbbc30811fe151992654cdc5fa62a9a0f4d0be/pylsl/pylsl.py#L635)

As I was editing the issue, I realized that it had to do with the timeout ('reconnecting' in the C++ error message was the clue).