labstreaminglayer/pylsl #96: Packaging improvements

mscheltienne, 2024-12-13:
Hi @cboulay! I see that you pushed quite a few nice changes. I have a couple of questions and tips for the packaging. I now see:

https://github.com/labstreaminglayer/pylsl/blob/fb0ae2ed8a0b6eab3b373a1d614e557ab04c7eec/pyproject.toml#L67-L68

Accompagnied with attempts to re-target wheels on windows 32 and 64. Unless mistaken, you are not building and packaging `liblsl` within `pylsl`, thus you should have a source distribution and a single binary distribution `pylsl-**-py3-none-any.whl`. Overall, you should be able to get rid of `setup.py` entirely and since you reverted to `setuptools`, specify the following `pyproject.toml`:

```
[build-system]
build-backend = 'setuptools.build_meta'
requires = ['setuptools >= 64.0.0']

...

[tool.setuptools]
include-package-data = false

[tool.setuptools.packages.find]
exclude = ['test']
include = ['src/pylsl*']
```

cboulay, 2024-12-13:
Hi @mscheltienne ,

I tried to explain it in #95 but I can see how that got lost with all the other spam this repo was producing yesterday thanks to me.

In short, I want the win_amd64 and win32 builds to include liblsl (lsl.dll) because Windows users have a tougher time installing liblsl on the PATH. However, for ALL OTHER PLATFORMS, I don't bundle liblsl because it's relatively simple to build/install -- the strategy instead is to provide a useful error message when liblsl can't be found. Back when we were bundling the binaries with pylsl, there was always a platform or 2 users had that weren't covered and it wasn't worth the effort to support them all.

A couple questions for you:

* Will non 3.10 python's using pip still pickup your wheels, given that they are tagged cp3.10?
* How do you build liblsl on Mac with code signing so it isn't rejected by OS security? Or do users have to rebuild liblsl every time they pip install pylsl?

mscheltienne, 2024-12-13:
I missed #95 entirely 🙈 Alright that makes sense. 

For `mne-lsl`, no the minimum supported version is 3.10. `mne` is following the guidelines from the scientific python community, [SPEC 0](https://scientific-python.org/specs/spec-0000/) recommends to drop python 3.10 in quarter 4 of 2024 (now) thus the next release will probably be tagged `cp311`. Same for `conda-forge` installs.

For macOS code signing.. good question and I don't have a definitive answer. I hope we are taking this into account (I'll have to check on a macOS laptop on Monday), my guess is that either `cibuildwheel` or `delocate-wheel` are taking care of it.

mscheltienne, 2024-12-13:
*The tag is `cp310-abi3`, thus anything above 3.10 is supported.

closed 2026-09-20 by cboulay