labstreaminglayer/pylsl #102: QUESTION: pylsl and liblsl version correspondence

sappelhoff, 2025-02-19:
Hi @cboulay 👋 

I see that recently there have been a few pylsl releases (now at 1.17.6).

The last time I updated pylsl, I was at version 1.16.2, and I have some logic built into my code that double checks that the liblsl version that I am using is ALSO at 1.16.2 https://github.com/sccn/liblsl/releases/

Now I happened to have updated pylsl and ran into an error where my code told me that my pylsl and liblsl are out of sync. Hence my message here:

Is it intended for liblsl and pylsl to have to correspond in terms of versions?

Or can I simply use liblsl 1.16.2 with pylsl 1.17.6?

What are the sort of checks I should make before pairing liblsl and pylsl versions?

Thanks!

cboulay, 2025-02-19:
Yes, you can use the latest pylsl with the latest liblsl.

When I started versioning pylsl, I used the same version number as liblsl. This was a mistake. We should be able to make pylsl releases independent of liblsl releases.

I guess we need some way to verify pylsl version <-> liblsl version compatibility. This could be documentation only, or it could be in code. Any suggestions?

sappelhoff, 2025-02-19:
Thanks for the quick response!

> I guess we need some way to verify pylsl version <-> liblsl version compatibility. This could be documentation only, or it could be in code. Any suggestions?

IMHO ideally this would be done via test suites that run in CI, and whenever something is pushed to liblsl that breaks pylsl, or vice versa, the maintainers would see and could either remedy the problem, or publish a list of which version does not work with which other version 🤔

closed 2026-09-20 by cboulay