labstreaminglayer/pylsl #104: PyLSL lacking binary in Linux

Kaszanas, 2025-05-24:
It seems to me like this was already mentioned here: https://github.com/labstreaminglayer/pylsl/issues/6

I have a project that defines the `pylsl` as a dependency in the `pyproject.toml` while on Windows there is no additional steps to install. It seems to me like Linux just does not pull any files along with such installation.

If possible, plese indicate the required steps towards Linux installs to use pylsl.
What are the ways of installing a working pylsl on Linux?

cboulay, 2025-05-24:
You are expected to install liblsl at the system level, or otherwise direct pylsl to your liblsl binaries with environment variables. If you tried to run pylsl and it couldn't located the liblsl binaries then it should have provided an error message with instructions. If not then I'd like to know about that so I can make sure the procedure is more evident in the future.

Kaszanas, 2025-05-24:
I've made a bash script that installs it for me at this point.

```bash
# Replace CODENAME with your Ubuntu codename (e.g., noble, jammy, focal)
CODENAME=$(lsb_release -cs)
ARCH=amd64

# Get the latest release tag from GitHub API
TAG=$(curl -s https://api.github.com/repos/sccn/liblsl/releases/latest | grep tag_name | cut -d '"' -f 4)

# Build the .deb filename
DEB="liblsl-${TAG#v}-${CODENAME}_${ARCH}.deb"

# Download the .deb package
curl -LO "https://github.com/sccn/liblsl/releases/download/${TAG}/${DEB}"

# Install the package
sudo apt install -y ./${DEB}

# Find the liblsl.so path
LSL_SO_PATH=$(dpkg -L liblsl | grep '/liblsl.so$' | head -n1)

# Export PYLSL_LIB for this shell session
export PYLSL_LIB="${LSL_SO_PATH}"
echo "PYLSL_LIB set to: $PYLSL_LIB"
```

Maybe someone will find this useful.
I am closing this as I seem to have resolve the issue.

closed 2025-05-24 by Kaszanas